knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #354: MEGA prints only half the size after install TMC drivers

SjoerdNLD, 2022-08-08:
### Bug Description

movement is only 50% of the expected move. 5mm instead of 10mm pressed on the display.

### Steps to Reproduce

1. Install MKS TMC2225 drivers
2. Flash MEGA_TMC firmware
3. restore defaults and store eeprom
4. cycle power and press move x axis 10mm

**Expected behavior:** move 10mm

**Actual behavior:** move 5mm

knutwurst, 2022-08-09:
Please see #335 .

(Translated: https://github-com.translate.goog/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/issues/335?_x_tr_sl=de&_x_tr_tl=en&_x_tr_hl=de&_x_tr_pto=wapp)

closed 2022-08-09 by knutwurst

SjoerdNLD, 2022-08-10:
Thanks for the help, I excuted this, and my mega works agian normal:
M92
  M92 X80.00 Y80.00 Z400.00 E98.00
ok
M92 X160 Y160 Z800 E98
ok
M92
  M92 X160.00 Y160.00 Z800.00 E98.00
ok