knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #355: [FR] Showing the "real" remaining time

PapeCoding, 2022-08-10:
### Description

Would it be possible to parse the line `;TIME:<print time in seconds>` from a gcode file that was exported with Cura and display the remaining time (from first command in print, so line `;LAYER:0`) somewhere in the UI?

### Feature Workflow

1. Slice model with Cura and put gcode on SD-Card
2. Printer executes gcode normally and parses the `;TIME:<print time in seconds>` line
3. Once the `;LAYER:0` is reached, the printer starts a timer and counts down the time and displays it in the UI

#### Additional Information

I guess this will be hard, as the UI is not customizable, if I remember correctly, but would be immensely helpful.
The estimation that Cura makes is quite accurate for me (ignoring heat-up times) and would be super helpful to see instead of e.g. the progress in % through the file. Maybe other slicers output similar lines or you could just make it a feature when Cura is used?

Additionally, it would be possible to create custom lines that can be parsed in the start and end gcode, by using the replacement patterns that slicers support (see [here](http://files.fieldofview.com/cura/Replacement_Patterns.html) e.g. for Cura).

PapeCoding, 2022-09-12:
Still relevant. Please do not close.

knutwurst, 2022-09-15:
Hi @PapeCoding ,
I didn't have the opportunity to further develop the firmware for some time. I'm looking into this, but to be honest this is a relatively low priority as there is still one critical bug and one feature that needs to be fixed.

But I'll remember that and put it in my backlog, ok?

You can either close this ticket or leave it open. But I wrote it down.

best regards.,
Olli

closed 2022-10-22 by github-actions[bot]