knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #357: Trying to locate configuration for specific build

EasyG0ing1, 2022-08-11:
Hello,

I've been using ```MEGA_X_DGUS_TMC_BLT_10_v1.4.4.hex``` and it's been working great. my new BLTouch works perfectly as does the replacement 2208 drivers.

However, I would like to enable higher resolution where bed leveling records more data points.

I followed your instructions [here](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/Howto:-Firmware-mit-PlatformIO-selbst-kompilieren), and I even managed to compile the same firmware, but I cannot find the specific ```Configuration.h``` or the ```Configuration_adv.h``` for this build so that I can enable higher resolution bed leveling. 

Can you point me in the right direction so I can enable that feature?

Thank you,

Mike Sims

Glutamat42, 2022-09-04:
They are inside the Marlin directory (Marlin/Configuration.h / Marlin/Configuration_adv.h)

EasyG0ing1, 2022-09-04:
@Glutamat42 
> They are inside the Marlin directory (Marlin/Configuration.h / Marlin/Configuration_adv.h)

I saw those, but I am assuming that there is a unique version of those files for each unique build of the firmware? Which is what I am specifically pursuing here ... or am I missing something? Because so far, I have not found a different Configuration.h / Configuration_adv.h for each different build of the firmware that is offered in the [release downloads](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/releases) of the repository.

Glutamat42, 2022-09-04:
As far as i understand it the differences between all available firmwares are defined via buildflags (ini/i3_mega.ini) and "combined" with Configuration.h
I used Marlin/Configuration.h to enable the 2nd serial port to connect an esp and it worked fine (confirming my understanding of the project structure)

EasyG0ing1, 2022-09-05:
@Glutamat42 
> As far as i understand it the differences between all available firmwares are defined via buildflags (ini/i3_mega.ini) and "combined" with Configuration.h I used Marlin/Configuration.h to enable the [...]

Thank you for confirming this. I realized that this is how it worked after my last post.

For anyone reading this in the future, where you might assume (as I did) that there is a unique configuration.h for each firmware build, this is how you properly customize your specific firmware build:

- Follow [these instructions](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/Howto:-Firmware-mit-PlatformIO-selbst-kompilieren). I highly recommend using Visual Studio Code with PlatformIO as recommended in the instructions, regardless of what your favorite IDE is, because it will make this whole process very simple.
- Use Configuration.h and Configuration_adv.h from the root Marlin folder in the project to implement your desired options for the firmware.
- Click on the little alien head and find your specific build (ex: MEGA_X_DGUS_TMC_BLT_10)
- Click on that build and VSCode will do some processing of some files. Wait till that is done (and that only happens once, it does not happen again when you make further changes to the config .h files).
- Under General in your firmware build, click on Build.
- There will be a file in the .pio/build/(whatever_firmware_option)/ folder named firmware.hex - this is the file you flash to your printer.

EasyG0ing1, 2022-09-05:
@knutwurst 
Als jemand, der SEHR neu auf diesem Niveau des 3D-Drucks ist, wo es eine Notwendigkeit werden kann, eine angepasste Firmware für einen bestimmten Drucker zu haben, war mir basierend auf dem Wiki nicht klar, wie ich die Konfigurationsoptionen in der Firmware ändern könnte für mein konkreter Aufbau. Daher empfehle ich Ihnen, dem Wiki einen kurzen Abschnitt hinzuzufügen.

Ich empfehle Ihnen, Folgendes zu [dieser wiki-seite](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/Howto:-Firmware-mit-PlatformIO-selbst-kompilieren) direkt nach Abschnitt 5 und vor Abschnitt 6 hinzuzufügen, um denjenigen in Zukunft zu helfen, die wie ich damit zu kämpfen haben:

#### Änderungen an Firmware-Optionen vornehmen
Das Ändern der Marlin-Firmware-Optionen ist eine einfache Sache der Bearbeitung von Config.h und Config_adv.h im Root-Marlin-Ordner in VCode. Nachdem Sie die gewünschten Änderungen vorgenommen haben, erstellen Sie die Firmware wie oben beschrieben. Die Root-Konfigurationsdateien werden auf die Build-Option angewendet, die Sie unter der Registerkarte „Alien-Kopf“ auswählen.

knutwurst, 2022-09-15:
Hi @EasyG0ing1 ,
Thank you very much for your feedback!
Yes, it looks like I should revise the guide again to make this clearer. I actually thought that point 7 and 8 (The examples) show exactly what you intended. Namely both changing a file - it doesn't matter whether it's pins_ramps.h or configuration.h. And just the selection of a profile. The text also says:

> A last important feature of this Hybrid Firmware are the profiles mentioned above, which are used to control the individual printer configurations. These definitions can be found in many places in the source code, often with a #IF ENABLED(<configuration name>). Various paths are switched on and off or linked to conditions at compile time.

I thought that was enough. But maybe I'll rewrite it again.

Thanks for the hint!

Best regards,
Olli

EasyG0ing1, 2022-09-16:
@knutwurst - What I can say is that after reading the instructions TWICE and being someone who had NEVER BEFORE compiled firmware for a 3D printer, I was still unclear about how the specific build profiles were connected to the configuration files. And I went into this ASSUMING that each different build profile had its own set of config files.

My assumption was based on what seemed obvious to me; that each build would naturally require different settings in configuration and configuration_adv

I understand NOW that my assumption was incorrect because the config files branch and select different options based on the flags that are set from a given build profile, but before I learned that, I simply assumed that there would be one set of config files for each build option and when I could not find those files, I created this issue, because the wiki didn't tell me how the builds get their configs either so it remained a large question mark in my mind.

Being a Network Administrator for 25 years now, and an occasional developer in Java and Arduino's, I am going under the assumption that I can't be the first person who was confused about this, and the odds are low that I will be the last one ... but we both see where assumptions have gotten me to this point... 

All I am suggesting is that something as simple as this, inserted into the wiki would prevent this confusion from happening to anyone else in the future.

> If you want to make changes to your specific printer build, simply make your changes in ```configuration.h``` and ```configuration_adv.h``` in the root Marlin folder. Then when you are done setting your preferred options, click on the alien head, then expand out your chosen build label, then click on ```build``` just underneath the label. The hex file will be placed in the ```.pio/build/YOUR_BUILD_OPTION``` folder.

It certainly cannot hurt to add more detail to the wiki in my opinion ... re-writing it entirely - I don't believe is necessary at all ... just adding those few sentences should eliminate any confusion in beginners moving forward.

EasyG0ing1, 2022-09-28:
@knutwurst 

I have taken a portion of the wiki and inserted my proposed VERY SIMPLE addition to the page and put it on my web server for you to review. The proposed change is highlighted in yellow.

Here is the [English version
](https://simtechdata.com/knutwurst/proposedchange.html)Here is the [German version](https://simtechdata.com/knutwurst/proposedchangegerman.html)

closed 2022-11-04 by github-actions[bot]