knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #360: [BUG] 32bit AI3M Printer no longer booting after firmware 1.4.4 flash

HB-Stratos, 2022-08-18:
### Bug Description

I flashed my stock, previously untouched 32 bit Anycubic I3 Mega with the 1.4.4 firmware, specifically file MEGA_v1.4.4.hex as I was told in the beginner documentation. To flash successfully I had to remove jumper JP1 and put jumper JW1 into the USB position. I found out about that [here](https://github.com/napyk/trigorilla-pro) after my original flash attempt told me that the bootloader was locked. Flashing was done via the octoprint flashfirmware plugin, using stm32 mode.
After flashing now the display no longer comes on, and the printer makes no noise. The small LED on the mainboard lights up however and my computer detects a connection, so the board seems to still be alive. 

### Steps to Reproduce

see above

**Expected behavior:** [What you expect to happen]

The printer should boot

**Actual behavior:** [What actually happens]

The printer stays dead, presumably unable to load and execute the code, so only the bootloader runs.

#### Additional Information

None

knutwurst, 2022-08-19:
I am not sure if you are serious. Since you obviously know that you have an STM32 processor and also think that your bootloader is still there, I assume that you should actually know what you are doing.

This, in turn, is in contrast to the fact that you did not read the [Beginner's Guide](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/Beginner's-Guide-(English)) once which clearly states a compatibility list.

Quote: "This firmware works on many Anycubic brand 3D printers, in which a "Trigorilla" mainboard with an Atmega2560 was installed."

As far as I can see, your location is also germany, so there is also a german Guide [here](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/Beginner's-Guide-(Deutsch)) which is also linked in every download section.

So you have no choice but to reinstall the original firmware.

HB-Stratos, 2022-08-19:
I did indeed miss that in the beginners guide. I am not sure why I did, but I missed that. I had originally come from https://github.com/davidramiro/Marlin-Ai3M-2.0.x here, where it states on the front page that at least the underlying marling 2.0.x supports 32 bit board, which had brought me to the assumption that all 2.0.x firmwares of marlin would work with my board. But it seems that assumption was false.
Are you aware of any Marlin firmwares for my printer? Is there any project out there that made a proper marlin firmware for the 32 bit anycubic?

knutwurst, 2022-08-19:
32 bits is not equal to 32 bits.
There is an (admittedly large) list of supported boards, but the Trigorilla Pro (that's what yours is called) isn't currently supported as far as I know.

The printer in its original state is also based on CHUTU and not on Marlin. In addition, the A4899 drivers are soldered and cannot be exchanged for silent steppers.

There may be a project for it, but unfortunately I don't know anything about it. According to Google there is a project for the Predator and on Cluts3d a Mega S project (closed source). But I can't say anything about that.

My suggestion would be to put in an original Trigorilla + TMC2208 and then put it on Marlin, or swap everything out and put it on RepRap or Klipper. But that's all quite a lot of work.

I'm sorry I can't help with that.

closed 2022-09-25 by github-actions[bot]