knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #365: Mesh Bed Leveling failed after FW update

quantenquark, 2022-09-10:
Hello, everyone
After the connector for the print head burned out, I connected the BLTouch via a JST connector on the mini circuit board and bridged the contacts. Runs perfectly.
I also flashed the appropriate firmware on Marlin 2.0.9.2 (MEGA_S_BLT_10_v1.4.4, the motor plugs were swapped after the TMC update). After flashing, FW defaults were loaded and saved.

Errors occur during commissioning with new mesh bed leveling.
The points I0-J0 to I4 J0 are cleanly approached and sampled. After that, the sensor moves to I4-J1, tests 4 or 5x without a result and then moves to I4-J2, tests several times again and then wants to move to a position on the right outside of the print bed. Here I have to stop manually to protect servos and parts.

I have the impression that the start coordinates of the printer are wrong, the X-Stop when the printer starts is not approached.
What am I doing wrong?Any help is welcome.

Attached the log after start:

Connecting to port /dev/ttyUSB0, baudrate 250000
Handshake attempt #1 with timeout 2.0s
Connected to: Serial<id=0xa8814df0, open=True>(port='/dev/ttyUSB0', baudrate=250000, bytesize=8, parity='N', stopbits=1, timeout=2.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Send: N0 M110 N0*125
Recv: start
Changing monitoring state from "Detecting serial connection" to "Operational"
Send: N0 M110 N0*125
Recv: Marlin 2.0.9.2
Recv: echo: Last Updated: 2022-05-12 | Author: (knutwurst)
Recv: echo: Compiled: May 12 2022
Recv: echo: Free Memory: 1584  PlannerBufferBytes: 1568
Recv: echo:V85 stored settings retrieved (692 bytes; crc 12805)
Recv: echo:; Linear Units:
Recv:   G21 ; (mm)
Recv: echo:; Temperature Units:
Recv: echo:  M149 C ; Units in Celsius
Recv: echo:; Filament settings (Disabled):
Recv: echo:  M200 S0 D1.75
Recv: echo:; Steps per unit:
Recv: echo:  M92 X80.00 Y80.00 Z400.00 E393.00
Recv: echo:; Max feedrates (units/s):
Recv: echo:  M203 X500.00 Y500.00 Z6.00 E40.00
Recv: echo:; Max Acceleration (units/s2):
Recv: echo:  M201 X3000.00 Y2000.00 Z60.00 E10000.00
Recv: echo:; Acceleration (units/s2) (P<print-accel> R<retract-accel> T<travel-accel>):
Recv: echo:  M204 P1500.00 R1500.00 T3000.00
Recv: echo:; Advanced (B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>):
Recv: echo:  M205 B20000.00 S0.00 T0.00 X8.00 Y8.00 Z0.40 E10.00
Recv: echo:; Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:; Auto Bed Leveling:
Recv: echo:  M420 S0 Z0.00 ; Leveling OFF
Recv: echo:  G29 W I0 J0 Z0.00000
Recv: echo:  G29 W I1 J0 Z0.00000
Recv: echo:  G29 W I2 J0 Z0.00000
Recv: echo:  G29 W I3 J0 Z0.00000
Recv: echo:  G29 W I4 J0 Z0.00000
Recv: echo:  G29 W I0 J1 Z0.00000
Recv: echo:  G29 W I1 J1 Z0.00000
Recv: echo:  G29 W I2 J1 Z0.00000
Recv: echo:  G29 W I3 J1 Z0.00000
Recv: echo:  G29 W I4 J1 Z0.00000
Recv: echo:  G29 W I0 J2 Z0.00000
Recv: echo:  G29 W I1 J2 Z0.00000
Recv: echo:  G29 W I2 J2 Z0.00000
Recv: echo:  G29 W I3 J2 Z0.00000
Recv: echo:  G29 W I4 J2 Z0.00000
Recv: echo:  G29 W I0 J3 Z0.00000
Recv: echo:  G29 W I1 J3 Z0.00000
Recv: echo:  G29 W I2 J3 Z0.00000
Recv: echo:  G29 W I3 J3 Z0.00000
Recv: echo:  G29 W I4 J3 Z0.00000
Recv: echo:  G29 W I0 J4 Z0.00000
Recv: echo:  G29 W I1 J4 Z0.00000
Recv: echo:  G29 W I2 J4 Z0.00000
Recv: echo:  G29 W I3 J4 Z0.00000
Recv: echo:  G29 W I4 J4 Z0.00000
Recv: echo:; Endstop adjustment:
Recv: echo:  M666 Z0.00
Recv: echo:; Hotend PID:
Recv: echo:  M301 P12.28 I0.75 D50.06
Recv: echo:; Bed PID:
Recv: echo:echo:  M304 P251.78 I49.57 D319.73
Recv: echo:; Z-Probe Offset:
Recv: echo:  M851 X-2.00 Y-25.00 Z-0.40 ; (mm)
Recv: echo:; Linear Advance:
Recv: echo:  M900 K0.00
Recv: echo:; Filament load/unload:
Recv: echo:  M603 L538.00 U555.00 ; (mm)
Recv: //action:prompt_end
Recv: ok
Send: N1 M115*39
Recv: FIRMWARE_NAME:Marlin 2.0.9.2 (knutwurst, Github) (May 12 2022 13:32:22) SOURCE_CODE_URL:github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S PROTOCOL_VERSION:1.0 MACHINE_TYPE:Knutwurst's Anycubic Printer EXTRUDER_COUNT:1 UUID:b6153152-86fb-4cd8-bda7-91c6a04c467d
Recv: Cap:SERIAL_XON_XOFF:0
Recv: Cap:BINARY_FILE_TRANSFER:0
Recv: Cap:EEPROM:1
Recv: Cap:VOLUMETRIC:1
Recv: Cap:AUTOREPORT_POS:0
Recv: Cap:AUTOREPORT_TEMP:1
Recv: Cap:PROGRESS:0
Recv: Cap:PRINT_JOB:1
Recv: Cap:AUTOLEVEL:1
Recv: Cap:RUNOUT:0
Recv: Cap:Z_PROBE:1
Recv: Cap:LEVELING_DATA:1
Recv: Cap:BUILD_PERCENT:0
Recv: Cap:SOFTWARE_POWER:0
Recv: Cap:TOGGLE_LIGHTS:0
Recv: Cap:CASE_LIGHT_BRIGHTNESS:0
Recv: Cap:EMERGENCY_PARSER:1
Recv: Cap:HOST_ACTION_COMMANDS:1
Recv: Cap:PROMPT_SUPPORT:1
Recv: Cap:SDCARD:1
Recv: Cap:REPEAT:1
Recv: Cap:SD_WRITE:1
Recv: Cap:AUTOREPORT_SD_STATUS:0
Recv: Cap:LONG_FILENAME:0
Recv: Cap:THERMAL_PROTECTION:1
Recv: Cap:MOTION_MODES:0
Recv: Cap:ARCS:1
Recv: Cap:BABYSTEPPING:1
Recv: Cap:CHAMBER_TEMPERATURE:0
Recv: Cap:COOLER_TEMPERATURE:0
Recv: Cap:MEATPACK:1
Recv: ok
Send: M21
Recv: echo:No SD card
Recv: //action:notification SD Init Fail
Recv: ok
Send: M420V1
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4
Recv:  0  =====  =====  =====  =====  =====
Recv:  1  =====  =====  =====  =====  =====
Recv:  2  =====  =====  =====  =====  =====
Recv:  3  =====  =====  =====  =====  =====
Recv:  4  =====  =====  =====  =====  =====
Recv: 
Recv: echo:Bed Leveling OFF
Recv: echo:Fade Height OFF
Recv: ok
Send: M851
Recv:   M851 X-2.00 Y-25.00 Z-0.40 ; (mm)
Recv: ok
Send: M155 S2
Recv: ok
Send: M876 P1
Recv: ok

>>>>Here I have to shut down the printer as the head moves out of spec.

quantenquark, 2022-09-10:
P.S. After downgrading the FW to 1.1.5 MBL failed after probing position I0 J2 too.

knutwurst, 2022-09-14:
This doesn't read like a firmware issue. If the 4 point easy leveling works, then the coordinates can't be wrong either.

Have you thought about loading and saving the defaults after changing the firmware? This is extremely important, otherwise there will be garbage in the EEPROM.

Otherwise I would check whether your motor drivers are still working properly, whether everything is running smoothly (and not stuck) and possibly whether the power supply unit is defective. Because that can also be the case.

It sounds like an electrical problem to me.

quantenquark, 2022-09-14:
Hi
As the printer is running with the originall FW too completing one of my standard prints successfully I would not expect any error with the servos, DC power unit or another component.
For sure, I did load the FW defaults and saved them to EEPROM.

Is there any chance to reset the printer itself to start with a "blank" system?

quantenquark, 2022-10-15:
Well, reading lots of posts I have found out that the x-stop is permanently triggered. There is also no sound with the Anycubic FW 1.1.5 while triggering the x-stop. y- and z-stop are giving the "beep".

Using command M119, this is the result:

Reporting endstop status
x_min: TRIGGERED
y_min: open
z_min: open
z2_min: open

Measuring the states with a multimeter the switch does work as expected. NO and NC are ok, no faillure.

There is no difference if the x-stop is activly activated or not as in NO and NC postition results in the ssame M119 check:
x_min: TRIGGERED

Any ideas?

quantenquark, 2022-10-17:
I have checked the cables, connectors and plugs again. No damaged pin, broken cable, etc.
BUT only as single wire and not while being connected to the board.
It turned out that x-pin measured against x+ pin allow a permanent voltage passage. 

At least one bug has been found! I will exchange the board and hope for success.

quantenquark, 2022-10-23:
Hi,
replacing the board has solved the issue.
Thanks for taking care.
73
Achim

closed 2022-10-23 by quantenquark