knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #367: firmware with laser engraving

djepsylon, 2022-09-11:
### Bug Description

Hi,
Sorry to disturb you in this section but impossible to ask you a question in an other way, the link you give for general help send on new issue choice where is only bug and feature request choices.
I have a mega pro with laser recently.
I wouldn't tell you anything by telling you that the stock firmware is more than badly made lol.
I've been looking for quite a while, without finding answers, for a way to compile the firmware with the built-in bitmap converter so as not to lose the laser . 
Anyway, I came across your github and I see that you have been working on the same issue. 
My questions are: 
  have you managed to make a firmware that allows you to keep the laser function without any modification of the machine? 
  Did you manage to find how to integrate the bmp converter or did you do otherwise? 
  And finally, would you agree to explain to me how you did it or provide me with a functional marlin?

Best regards

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

djepsylon, 2022-09-11:
Hi,
Sorry to disturb you.
I have a mega pro with laser recently.
I wouldn't tell you anything by telling you that the stock firmware is more than badly made lol.
I've been looking for quite a while, without finding answers, for a way to compile the firmware with the built-in bitmap converter so as not to lose the laser . 
Anyway, I came across your github and I see that you have been working on the same issue. 
My questions are: 
  have you managed to make a firmware that allows you to keep the laser function without any modification of the machine? 
  Did you manage to find how to integrate the bmp converter or did you do otherwise? 
  And finally, would you agree to explain to me how you did it or provide me with a functional marlin?

Best regards

knutwurst, 2022-09-14:
Hi,
so currently the laser only works with LightBurn via USB, but @stklcode is already working on implementing the laser functionality via the SD card to print BMP files.

Unfortunately I don't own a Mega Pro myself so it's hard for me to replicate these features, but @stklcode is more than capable of doing it and has done a great deal for this project.

In other words: be patient :)

Best regards,
Olli

djepsylon, 2022-09-15:
hi,
thank you for your response.
If the laser can work with light burn in USB, I prefer than the stock way.
There is no physical modification to do, just the firmware? Can you give me
a link to the firmware that I can use (if possible marlin 2.1.1 or bugfix
2.1.1) and explain to me how to use lightburn?

Thank you very much in advance
Best regards.

Godefroid Philippe

djepsylon, 2022-09-15:
If i undurstand well, i must take the  "MEGA_P_DGUS_v1.4.4.hex" firmware if i don't have bltouch and i must take 
" MEGA_P_DGUS_BLT_10_v1.4.4.hex  " ifi have motherboard trigorilla 1.0 or " MEGA_P_DGUS_BLT_11_v1.4.4.hex "  firmware if i have bltouch and motherboard trigorilla 1.1, correct?

now how can i use the lazer with lightburn?

stklcode, 2022-09-16:
> If i undurstand well, i must take the "MEGA_P_DGUS_v1.4.4.hex" firmware if i don't have bltouch and i must take " MEGA_P_DGUS_BLT_10_v1.4.4.hex " ifi have motherboard trigorilla 1.0 or " MEGA_P_DGUS_BLT_11_v1.4.4.hex " firmware if i have bltouch and motherboard trigorilla 1.1, correct?

Correct.

> now how can i use the lazer with lightburn?

Add a new machine, select "Marlin" flavor and add some basic values.

As with 3D printing, this is needs some practice and experimentation. Not necessarily a 1-click-thing.

[image]
[image]
[image]

I personally prefer to do homing and focusing (different material thickness needs different height) manually, so it's commented out in the GCode sequence. If you know your parameters. The screnshots are outdated, nowadays I use a different sequence with air optional assistance and case lighting - but this required hardware modifications. What is shown above should work fine without any mods.

djepsylon, 2022-09-16:
ok.
a big big thanks for your perfect response!
i flash the firmware and i ask you the results.
if i have some questions can i ask you?
and for use the laser with laightburn, i must do like stock method? i
disconnect the hotend and connect the laser?

djepsylon, 2022-09-17:
just another question, is there a 2.1.1 marlin's version of that firmware?
this is the 2.0.9.3 this one,right?

knutwurst, 2022-09-17:
> and for use the laser with laightburn, i must do like stock method? i
disconnect the hotend and connect the laser?

That's correct.

>just another question, is there a 2.1.1 marlin's version of that firmware?
this is the 2.0.9.3 this one,right?

The new 2.1.1 version is already on the way, thanks to @stklcode , but there is currently no public release. You can build it by yourself, but the new version also needs a bit more testing.

The current stable 1.4.x is based on Marlin 2.0.9.2.

djepsylon, 2022-09-17:
okay thanks a lot

djepsylon, 2022-09-18:
> another thing,for flashing the firmware i must do only with cura?

i ve another problem i ve the mainboard 0.0.2 in my mega pro not the 1.0 or
1.1 , what firmware i must take?

djepsylon, 2022-09-18:
[image]

stklcode, 2022-09-18:
https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/Beginner's-Guide-(English)#trigorilla-002

> The third version of the Trigorilla mainboard was installed until mid-2020. It can be recognized by the missing logo and the version number **0.0.2** and by the large green terminals for the power supply. Here the pin assignment for the servo connection was set back to the original one, so that this board **behaves like a Trigorilla 1.0**.

djepsylon, 2022-09-18:
Wow lolll sorry I don't remember that info. Thanks to remember me.

Godefroid Philippe
EPSYLON 3D

djepsylon, 2022-09-19:
hi,
I've flashed the printer and it works correctly and i have mounted a bltouch too.
I'm going to try the laser function with lightburn after a good sleeping time lolll.
I tell you the results of my tests.
for now the printing quality was very very better than stock.
thanks a lot again for your help.

djepsylon, 2022-09-20:
hello again
I managed to make the laser work with difficulty only there are problems I
have to put in lightburn "M03/M05" so that the laser lights up but it seems
that in this mode it only goes to 100% or 0% because I tried to print a
number "1" completely black and it only gives me the outlines while in the
preview it should have been black and I had put "infill" in the layers and
if I put "on line" the nothing happens and by putting "M106/M107" there the
movements are made but the laser does not turn on. then there is the point
of origin which looks strange the surface looks offset compared to the real
surface I understand that it is due to the laser which is hung on the head
and that this causes an offset but how to settle this?
sorry to ask questions may be stupid in your eyes but I completely discover
the laser function.
in short, how to operate the laser correctly?[image:
WM-Screenshots-20220920045751.png]
[image: WM-Screenshots-20220920053206.png]
[image: WM-Screenshots-20220920053020.png]
[image: WM-Screenshots-20220920053037.png]
[image: WM-Screenshots-20220920053140.png]
[image: WM-Screenshots-20220920053154.png]

stklcode, 2022-09-20:
Sound more like a question to the _LightBurn_ community. Has nothing to do with the printer firmware. Most of the following is answered in their documentation: https://lightburnsoftware.com/pages/lightburn-documentation

You can (and must) configure every layer individually whether the outline should be engraved (thickness, intensity, number of passes) and/or the content (intensity, number of passes).

You have to configure laser offsets somewhere in the machine settings. I is around -75mm in Y-direction.
(see screenshots above) The G-Code initialization may be extended to your individual needs.

If you want “zero“ to be ad the corner of your target object, just move there manually and apply “G92 X0 Y0“. If you want „zero“ to be the bottom left corner of the hotbed,  you may use homing (“G28“ or “G28 X0 Y0“ for X/Y only) and apply “G92 X0 Y-75 Z0“ or “G92 X0 Y-75“ respectively.

As said, it’s is not trivial to get good results on first shot.

djepsylon, 2022-09-21:
I've read all the instructions and I don't find any answer to my questions. The problem is that I can't get the burner to work properly. Either it passes through the black and hardly marks the lighter parts of an image or vice versa. For example I am trying to engrave the anycubic lodge and the dark parts cut through the craft paper and the lighter parts can hardly be seen if at all.
What do you recommend for me as setting speed, power, distance z, etc? is it correct to have it in order M03/M05?
I no longer understand anything. This lazer worked great before and now there is no way to do anything with it. On other hand, the printing works great. I really need help.

[image]

stklcode, 2022-09-21:
I'm sorry there is no straight-forward one-click recipe.

I have used extremely different settings depending on the material (paper, cardboard, wood, plastic, ...) and the actual task (engraving, cutting, melting, ...)

This was my very first "productive" work. Self-adhesive craft paper. The label is about 30x20mm in size. 

[image]

Large text: fill, 10mm/s, 48% (not cut through, just "burned paper")
Small text: line, 10mm/s, 48%
Symbols: line, 10mm/s, 45%
Outline: line, 10mm/s, 75%, 3 passes

You really have to figure out which values work for your task. In the above example, using 48% for the symbols was just cutting while 45% was fine. Just trial&error, made about 10 pieces until the result was OK.

If you have outlines to be cut, you should laser the text first and cut in the very end. Otherwise you might blow away the cut-out piece of paper easily.

Also make sure, the focus is correct. for cutting it should be really focused, but remember to screw the laser module down and place the printer on a solid ground. Otherwise you might see some shaking. If you just want to "paint" you might use some off-focus as well. E.g. if you want some thick lines, focus out 0.5 to 1mm and ramp up the power accordingly.

[image]

This was my first show on PCB lasering (copper with photo-sensitive layer). Didn't use the screws and too tight focus. Laser power was 31%, structures are <0.5mm

djepsylon, 2022-09-21:
hi, thanks a lot for all responses.
Well that's the problem, I can't find where the function is where you cut
and where you just engrave. and it seems that below 55 or 60% nothing
happens and I redid my focus with black paper (easier to see) it is
impeccable so I have trouble understanding. in addition, the speeds are in
mm/min and not in mm/sec. i'm at 400mm/min its too fast if i understand
well.
wher is the setting for cut or engrave?

[image: WM-Screenshots-20220921214258.png]

djepsylon, 2022-09-21:
you have not the same laser than with the printer? you have another one?

stklcode, 2022-09-22:
Every word I wrote in this context targets the default Anycubic i3 Mega Pro laser module that came with the printer. Number 4007777956, 1500mW diode Laser with approx. 450nm wavelength. The above examples were made with some Knutwurst 1.4.x firmware.

I don’t know if there is any spread, tolerance or similar with the modules, I do have exactly one, so I cannot compare anything.

djepsylon, 2022-09-22:
OK
I'm starting to understand apparently, the laser is not controlled in the
same way with lightburn and the original converter from anycubic. It would
seem that for a job requiring 15% power with the anycubic bmp converter
requires 55 or even 60% power with lightburn. it would also seem that
@stklcode is working on a version with the bmp converter like the original
version....
That would be great because I can no longer do without the print quality of
the @sklcode firmware, too bad I have so many problems with the laser....

stklcode, 2022-09-23:
> OK I'm starting to understand apparently, the laser is not controlled in the same way with lightburn and the original converter from anycubic. It would seem that for a job requiring 15% power with the anycubic bmp converter requires 55 or even 60% power with lightburn.

Right, 15% in die firmware feature does _not_ mean 15% laser power. The original firmware calculates the power based on average gray level of the bitmap pixels and adds some factor on top. They also use some custom on/off pseudo-PWM implementation, not exactly the linear ramp used by M3/M4 and, if I remember correctly, they have fixed speed.

> That would be great because I can no longer do without the print quality of the @sklcode firmware

Thanks, but it’s not “my“ firmware. Most work was done by @knutwurst and many others, either providing code or feedback to fine tune the parameters. Not to mention the Marlin community where our features come from.

I am close to finally port the BMP feature. Didn’t find much time last week though. Indication works fine (moving around the image border at 5% power), resolution and size detection works fine, but there is some trouble with the pixel to laser mapping left, so the actual print job does strange things at the moment.
I am going to migrate the custom implementations to use Marlin features and translate the BMP to proper GCode commands. The original code is quite hacky here and Marlin 1 structures are a little different.

djepsylon, 2022-09-25:
Okay thanks to all of yours far all that good work! 👍💪
Now we'll, I understand how It so difficult to have good results with lightburn. I suppose this is du to the very slow power of the laser unit?
i've found imagetogcode converter while waiting for you to finish the version of the firmware with converter and it is a little easier to use but I have a funny effect on the engravings, it only engraves the contours like in lightburn and if I increase the power it engraves inside the shapes but cut out the outlines which I don't want, do you have any idea?
Does it work better with a 5w unit? 
And finally I suppose the laser petformances are going to be better when you have finish the bmp converter or marlin converter.
How many times do you think you can publish the new firmware with bmp converter?

stklcode, 2022-09-25:
> How many times do you think you can publish the new firmware with bmp converter?

It’s finished, whenever it’s finished. The project has absolutely no priority at all to me, simply because I personally have no need for this feature and probably will never use it myself. Maybe a week, maybe month, maybe a year…

> And finally I suppose the laser petformances are going to be better when you have finish the bmp converter or marlin converter.

I don’t know what you are expecting of this feature. My best expectations are that is will become about as good as the original Anycubic feature - which I only used once and found it pretty poor. It’s a very basic BMP to laser power converter, far worse than any tool I’ve seen so far (LightBurn does a better job).
It also does not chance your environment. You still have a 1.5W diode laser, you still have to focus manually and choose some parameters.

djepsylon, 2022-09-25:
okay
thanks again for all your answers and sorry if my questions are strange to your eyes but i discover the laser universe lolll.

djepsylon, 2022-10-11:
Okay thanks to all of yours far all that good work! 👍💪
Now we'll, I understand how It so difficult to have good results with
lightburn. I suppose this is du to the very slow power of the laser unit?
Does it work better with a 5w unit?
And finally I suppose the laser petformances are going to be better when
you have finish the bmp converter or marlin converter.
How many times do you think you can publish the new firmware with bmp
converter?

A big thanks again
Best regards

Godefroid Philippe
EPSYLON 3D

closed 2022-11-17 by github-actions[bot]