knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #368: [FR] Offering a BLTouch Calibration Worksheet

EasyG0ing1, 2022-09-12:
I created this BLTouch Calibration worksheet that you might want to add to the wiki. It goes into much more detail about what BLTouch calibration is, and I think it would be a good document for beginners to have to both understand the process and have a worksheet to work from while going through the process.

[BLTouchCalibrationWorksheet.pdf](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/9546482/BLTouchCalibrationWorksheet.pdf)

I even converted it into German using Google Translate 

[BLTouchArbeitsblattDeutschKalibrieren.pdf](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/9546582/BLTouchArbeitsblattDeutschKalibrieren.pdf)

JernejL, 2022-09-28:
Can you also also add some guide & instruction on how to set bltouch's magnetic screw? because that affects the whole calibration and repeatability of the whole process.

EasyG0ing1, 2022-11-01:
> Can you also also add some guide & instruction on how to set bltouch's magnetic screw? because that affects the whole calibration and repeatability of the whole process.

I took a look at this and talked with several people who are well-versed with the BL Touch and the consensus seems to be unanimous that adjusting the magnetic screw would never be something done as a part of the normal setup process. And that if you're at the point where you are adjusting that screw, you have bigger problems than just getting things setup.

JernejL, 2022-11-10:
If you break the probe or want better precision, a procedure for screw adjustment is needed.

EasyG0ing1, 2022-12-12:
@JernejL I understand, but that is not part of the initial setup/configuration of the BLTouch, which this issue is precisely about.

closed 2023-01-19 by github-actions[bot]