knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #369: BLTouch should acting as Z-Endstop

t-poeschl, 2022-09-13:
Hello Together!

I need help with my Anycubic i3 Mega Pro.

I have installed Marlin, with a BLTouch, had removed the Z endstop at the time and can't find it, as I was told it was not needed. Today I found the time again and still have the problem that it tries to operate the Z endstop through the Knutwurst Marlin version - but I want the BLTouch to act as the endstop.

Can someone be helpful to me there?

Kind regards
Tobias

knutwurst, 2022-09-14:
Hi @t-poeschl ,
no, there is no preconfigured version which has this feature, because the dual z axis leveling is slow af and you need to adjust the bed every single time before printing. This is why I never included this feature. Also the limit switches are indeed more accurate compared to the bltouch probe.

But, if you like, take a look at this: https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/pull/61/files#diff-bb298d18dac4268225a7a53a5ad24844472530bdc63f6a98a2ba8d8158b75080

As you can see, it's not much at all. All you need to is copy&paste the small changes in the configuration.h which work for your printer. Also you need to set the probe pin accordingly in the pins_ramps.h. That's all.

A manual how to compile the firmware by yourself can be found in the wiki.

Best,
Olli

t-poeschl, 2022-09-14:
Good morning,

thank you for the quick feedback, then I will try it once so.

One question I still have, you said Probe Pin - Since I'm not so fit in the field, which pin would that be, according to the instructions as you have connected the BLtouch in the manual?

Kind regards
Tobias

t-poeschl, 2022-09-14:
I wont get it..

`In file included from Marlin\src\feature\bltouch.cpp:34:0:
Marlin\src\feature\bltouch.cpp: In static member function 'static bool BLTouch::triggered()':
Marlin\src\feature\../module/probe.h:43:55: error: 'Z_MIN_PROBE_ENDSTOP_INVERTING' was not declared in this scope
   #define PROBE_TRIGGERED() (READ(Z_MIN_PROBE_PIN) != Z_MIN_PROBE_ENDSTOP_INVERTING)
                                                       ^
Marlin\src\feature\bltouch.cpp:94:36: note: in expansion of macro 'PROBE_TRIGGERED'
 bool BLTouch::triggered() { return PROBE_TRIGGERED(); }
                                    ^~~~~~~~~~~~~~~
Marlin\src\feature\../module/probe.h:43:55: note: suggested alternative: 'Z_MIN_ENDSTOP_INVERTING'
   #define PROBE_TRIGGERED() (READ(Z_MIN_PROBE_PIN) != Z_MIN_PROBE_ENDSTOP_INVERTING)
                                                       ^
Marlin\src\feature\bltouch.cpp:94:36: note: in expansion of macro 'PROBE_TRIGGERED'
 bool BLTouch::triggered() { return PROBE_TRIGGERED(); }
                                    ^~~~~~~~~~~~~~~
Compiling .pio\build\MEGA_P_DGUS_TMC_BLT_11\src\src\feature\stepper_driver_safety.cpp.o
Compiling .pio\build\MEGA_P_DGUS_TMC_BLT_11\src\src\feature\tmc_util.cpp.o
*** [.pio\build\MEGA_P_DGUS_TMC_BLT_11\src\src\feature\bltouch.cpp.o] Error 1
=========================================================================================================================== [FAILED] Took 4.96 seconds ===========================================================================================================================

Environment             Status    Duration
----------------------  --------  ------------
MEGA_P_DGUS_TMC_BLT_11  FAILED    00:00:04.959
===================================================================================================================== 1 failed, 0 succeeded in 00:00:04.959 ===================================================================================================================== 

 *  The terminal process "C:\Users\Tobias Poeschl\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'MEGA_P_DGUS_TMC_BLT_11'" terminated with exit code: 1. `

knutwurst, 2022-09-14:
I replied to you on Facebook. Since this isn't a bug, I'm closing the ticket as well

closed 2022-09-14 by knutwurst