knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #381: [FR] (feature request title)

chevroux, 2022-10-25:
Sorry for not posting in the facebook group, but I don't have facebook.

I installed bltouch successfully and calculated the Z offset and set it with gcode.

Do I have to set the calculated offset in the on screen menu of the printer as well? Or is that offset added/subtracted from the offset set by gcode?

Thanks!

knutwurst, 2022-10-25:
Hi,
the offset-menu was introduced a long time after I wrote the installation instructions, but thanks for the hint :) 

The displayed offset is the current offset which is also set by the gcode. But it might not be refreshed if the printer is powered on and the offset changed via usb while displaying it. If you close the menu (via exit), change the offset by gcode and then re-open the menu, it should display the correct current offset.

chevroux, 2022-10-25:
Ah, I see! Thanks a lot for the quick answer!

closed 2022-10-25 by chevroux