knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #385: Anycubib I3 mega no way to disable runout sensor

effeerre75, 2022-11-06:
Form the "Special-menu" I choose the "disable fil. sensor" option, I hear the jingle and then save the EPROM settings.
When I come back to the main menu, at the bottom right always appear the error message "lack of filament"
Try with and without sensor connected. 
Triend alos to disable with M412 command via serial interface but it report that the command is unknown.
For now the only workaround I've found is to disconnect the sensor and short the two pin with a pice of wire.

Firmware is Marlin-2-0-x-Anycubic-i3-MEGA-S-master downloaded from this repo.

HaGueRe, 2022-11-20:
Why not placing a little piece of filament into the sensor? You will never see this message again.

effeerre75, 2022-11-20:
Because sensor operation is no longer reliable.

Mauker1, 2022-11-27:
I'm experiencing the same issue. My sensor is no longer working as expected. Disabling it would be ideal. But I get the same problem as OP.

Alkerion, 2022-12-02:
Easy solution : Just remove the sensor and cable, and put a jumper where the cable is connected.

Mauker1, 2022-12-02:
Yeah, that's pretty much what I ended up doing.

effeerre75, 2022-12-02:
hardware solution to fix a software problem… that’s pure microsoft way! 😃

aurarius1, 2022-12-10:
Maybe a dumb question but where do I need to put to jumper on the mainboard/hub?  As I connected my runout sensor to my raspberry pi running octoprint and the disable option is not working for me as well 😅

effeerre75, 2022-12-10:
I’ve just disconnected the cable from the sensor and put a bent there! 😃 

[image]

aurarius1, 2022-12-10:
Thanks, I'll do that then! 😃

Alkerion, 2022-12-10:
You can also do that, no more cable, clean.

[image]

aurarius1, 2022-12-10:
But then I'd need to replace the the connector as the remaining four cables need to be connected right?

Alkerion, 2022-12-10:
> But then I'd need to replace the the connector as the remaining four cables need to be connected right?

You're right, my mistake, I've tweaked this printed a lot that I don't even remember what was connected here...

closed 2023-01-17 by github-actions[bot]