knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #388: Enhance print stop sequence

anno0, 2022-11-08:
### Description

What about the stop button in the print menu?

I use FW HW-1.4.4 with a Mega X printer. When I press the stop button, the plate goes to the back. In my option it would be better that the plate goes to the front to present the part. Here I would prefer the end G-Code of Cura.

### Feature Workflow

1. Print something
2. Press stop button on LCD Display
3. The following G-Code could be stepped through

M104 S0                                    ; Extruder off
M140 S0                                    ; Heatbed off
M107                                       ; Fan off
G91                                        ; relative positioning
G1 E-5 F300                                ; retract a little
G1 Z+10 E-5 ; X-20 Y-20 F{travel_xy_speed} ; lift print head
G28 X0 Y0                                  ; homing
G1 Y290 F2000                              ; present the part
M84                                        ; disable stepper motors
G90                                        ; absolute positioning
M300 S440 P200                             ; Make Print Completed Tones
M300 S660 P250                             ; beep
M300 S880 P300                             ; beep

anno0, 2022-12-10:
up

sboehle, 2022-12-10:
I changed the following in configuration_adv.h:
#define EVENT_GCODE_SD_ABORT "G92 E0\nG1 E-5 F1000\nG28X\nG1 Y175 F3000"
But it is limited what can be entered here. When it is too much, the firmware can't be flashed to the printer anymore. I had this issue yesterday. The following example is **not** working:
EVENT_GCODE_SD_ABORT "G92 E0\nG1 E-5 F1000\nG28X\nG1 Y175 F3000\nM104 S0\nM140 S0\nM107\nM84"

closed 2023-01-17 by github-actions[bot]

anno0, 2023-01-17:
Can it marked as resolved?