knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #389: Out of border detection

anno0, 2022-11-09:
### Description
Is there any  restriction in the firmware, if a position is out of border?
As I attached, my MEGA X has a print room of 300 x 300 x 305 mm.

### Feature Workflow

1. Send the following G-Code via USB
 G21                                        ; metric values
 G90                                        ; absolute positioning
 G1 X400 Y400 Z400              ; out of border absolute positioning       

2. My MEGA X started positioning, when I just used the following command G1 X1600 

The question is, if it will crash?

anno0, 2022-12-14:
up

knutwurst, 2022-12-23:
> The question is, if it will crash?
No, it won't. 
It will move to the max position and stay there.

closed 2022-12-23 by knutwurst