knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #391: [BUG] BL-Touch z offset

The-Man-Is-Me, 2022-11-22:
I have looked at the wiki and this is indeed a bug. I am using the mega s.

When following the wiki tutorial of how to install a bl touch it instructs you to lower the bl touch till it hits the bed, however, when auto leveling it only touches the bed and does to retract in. I believe this is making the z offset completely wrong. 

Just follow the guide to reproduce this

I expected my z offset to be correct, but it was not.
I have followed other z offset guides, but they all require the bl-touch to act as the z probe, but that is not the case with this firmware.

knutwurst, 2022-12-17:
Which firmware version do you use?
This might be a Marlin-related bug. I'm on it.

The-Man-Is-Me, 2022-12-22:
Sorry for the late reply, I think you might be right on it being a marlin bug. M401 S1, or high speed mode is enabled by default for the MEGA_S_DGUS_BLT_10_v1.4.4.hex. I tried disabling it with M401 S0 but it made the auto leveling stop working all together.

knutwurst, 2022-12-22:
The issue is fixed in c03aa9d. It was just a small change in the marlin default values which had a huge impact. The fix will be deployed with the next release version.

The-Man-Is-Me, 2022-12-22:
Thank you, I really appreciate it.

The-Man-Is-Me, 2022-12-22:
Just as a question when will this next version be released, wondering if I should just wait or switch to mesh bed leveling for a bit.