knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #392: [Question] Are TMC 2208 v1.2 drivers compatible with linear advance?

meltiseugen, 2022-11-24:
I didn't find the "support" in the issue type, so I don't really know where to post a question tlpther then here

Alkerion, 2022-12-08:
Use TMC2209 instead and it will works flawlessly.

knutwurst, 2022-12-17:
Yes and no. 
The drivers are indeed compatible, but require really good cooling and a high Vref on E0.

meltiseugen, 2022-12-26:
> Use TMC2209 instead and it will works flawlessly.

@knutwurst do I need to make a custom build for the 2209? Or is it compatible with your TMC builds?

Alkerion, 2022-12-27:
2209 run cooler than 2208, they support more features like LA, so the choice between the two is quickly made.

I use 1.15V and it run flawlessly since months.

Just update your configuration.h and compile.

#if ENABLED(KNUTWURST_TMC)
  #define X_DRIVER_TYPE  TMC2209_STANDALONE
  #define Y_DRIVER_TYPE  TMC2209_STANDALONE
  #define Z_DRIVER_TYPE  TMC2209_STANDALONE
  #define E0_DRIVER_TYPE TMC2209_STANDALONE
  #define E1_DRIVER_TYPE TMC2209_STANDALONE

Boostie76, 2023-01-04:
> Yes and no. The drivers are indeed compatible, but require really good cooling and a high Vref on E0.

What do you mean by really good cooling? Is the stock fan enough? If not, should I disable LA again?

closed 2023-02-09 by github-actions[bot]