knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #396: Bltouch probe outside plate

leonel85, 2022-12-19:
### Description

Bltouch first point is outside the plate.
Is it possible to add an offset for the bltouch?
Not the distance between bltouch and nozzle. That’s ok

### Feature Workflow

1. Calibration of bltouch X,Y,Z offset
2. Run bed levelling

#### Additional Information

* Provide pictures or links that demonstrate a similar feature or concept.

leonel85, 2022-12-19:
I solve it with  PROBING_MARGIN_FRONT

`
#if ENABLED(KNUTWURST_BLTOUCH)
  #if PROBE_SELECTED && !IS_KINEMATIC
    #define PROBING_MARGIN_LEFT  10
    #define PROBING_MARGIN_RIGHT 10
    #define PROBING_MARGIN_FRONT 15   //increase this value
    #define PROBING_MARGIN_BACK  10
  #endif
#endif
`

closed 2022-12-19 by leonel85

knutwurst, 2022-12-20:
I think you've just shifted the problem, not solved it. If you set the X and Y offset (with M851) correctly, the exact points that the nozzle normally reaches will be sampled. There is usually no need to change the margin.

leonel85, 2022-12-20:
My home position is a little bit outside the plate. I measure the distance between the nozzle its  (28,17)
I see that default PROBING_MARGIN_FRONT 10 it's too close to the front border and back I have more free space (it's not real outside as I write).

may I  ask something related :  I see that when z go home position it reach the endstop a little bit down than the plate and goes a little bit up to home position. 
When it goes down the side of nozzle (the cone) hit the bed and the hot end flexes a little bit after that goes up to home position and everithing is fine.
I think that I have to push a little bit the bed backwards to avoid yhis situation but I cannot move the sensor or the bed mybe I have to add an extra offset.
I see it only because I was loking to the nozzle to see if bltouch offsets where correct

Thanks

leonel85, 2022-12-20:
the point of the nozzle is outside the bed but it is very close to the bed. When the nozzle goes down (a little bit down from the bed) to reach the endstop switch, the side of the nozze "push the bed back if the motor is unbreak, if the motor is alive the nozzle flex"
It's a very small flex . I think that I have this problem also in the past but i didn't realize it before.

Yes it stops when hit endstop switch and goes a little bit up to zero position.

I use the current branch firmware but I think it was the same also with old version

leonel85, 2022-12-20:
I think there is something wrong in the new firmware update.
I test the z end stop with M119. Both works perfect and are alined. I downgrade the firmware with latest release (MEGA_X_BLT_11_v1.4.4)
It works and the Behavior is different

As soon as it reach the sensor z axis goes up and goes down to the same position. So don't goes under the bed.
With this release the z zero is the point that it touch the sensor 

Current branch instead
As soon as it reach the sensor z axis goes up and that is the zero point. So goes under the bed to reach the sensor and after that goes up to zero point.
With this release the z zero point is not the as when it hit the sensor.

It could be that I compile something wrong. Could you compile for me MEGA_X_BLT_11 please

Thanks

knutwurst, 2022-12-20:
I am a little confused. Which version has which behavior?

The 1.4.4 (based on Marlin 2.0.9) is ok?
The 1.5.0 (based on Marlin 2.1.1) is not ok?
The current master (based on Marlin 2.1.2) is different again?

Could you make a video?
It's just hard for me to understand right now. But I will try to find the time to test all versions. Maybe there is a major issue...

leonel85, 2022-12-20:
The 1.4.4 works , 2.1.1 and 2.1.2 doesn't
I think that I find it 
#define HOMING_BACKOFF_POST_MM { 2, 2, 0 }    
it was 2,2,2 that's why goes up after touch the sensor and zero point should be 2 mm upper than sensor

knutwurst, 2022-12-21:
Wait, what?!
You are right! Thank you so much for that! 
It should definitely be `HOMING_BACKOFF_POST_MM { 2, 2, 0 } ` when not using the CHIRON environment!

leonel85, 2022-12-21:
Hi
Sorry, I don't know if this is the right place but I have some questions about the firmware and I have a trouble with the hotend. My printer is not heating the head anymore I have this error

> Send: M140 S60
> Recv: //action:notification Bed Heating...
> Recv: ok
> [...]
> Recv: Error:Heating failed, system stopped! Heater_ID: E0
> Recv: 
> Changing monitoring state from "Operational" to "Error"
> Send: M112
> Send: N2 M112*35
> Send: N3 M104 T0 S0*34
> Send: N4 M140 S0*97
> Changing monitoring state from "Error" to "Offline after error"
> Connection closed, closing down monitor

I try with another hotend but it doesn't work.
You think that it could be related to the firmware ?
There's something that I can debug to understand what is happen ?

leonel85, 2022-12-21:
There was an electrical problem on the board. I change the connector and solder new one
[image]

whc2001, 2022-12-29:
Just recompiled the newest firmware and the Z homing backup went away. The commit history brought me here and I'm a little bit confused. Is this function affecting auto leveling when enabled? I am currently using the original Z endstop switches and I've heard that this function would protect the switches from being depressed for a long time when not using.

leonel85, 2023-01-01:
Hi
 This function could be ok to protect the switch but it means that zero point is always upper than the limit switch so it could be used only if nozzle is out of the plate otherwise the nozzle will hit the table to reach the limit switch