knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #411: [BUG] Mesh Leveling moving down no more than 0.2 mm

mjaschen, 2022-12-27:
### Bug Description

When using the mesh-leveling feature, it's not possible to move the head down more than 0.2 mm (two times 0.1 mm).

### Steps to Reproduce

1. Use 4-point-leveling for rough adjustment
2. Start mesh-leveling
3. Move down by 0.1 mm → works, move down by 0.1 mm → works, move down by 0.1 mm → nothing happens

**Expected behavior:** After moving down on the z-axis by 0.1 mm twice it should be possible to move further down on the z-axis.

**Actual behavior:** After moving down on the z-axis by 0.1 mm twice all further movement commands are ignored.

#### Additional Information

In an older version it was possible to move down on the z-axis for more than 0.2 mm (last built version was from february 2022, I think it was 1.4.3):

```
Marlin 2.0.x
echo: Last Updated: 2022-02-18 | Author: (knutwurst)
echo: Compiled: Feb 21 2022
```

Why do you want to move down more than 0.2 mm? Because the bed is quite curved and around 0.25 mm offset is needed at the edges, see attached image.

[image]

anno0, 2022-12-27:
Have you found the Special menu in firmware 1.4.3? I have installed firmware 1.4.4 and have never seen this menu when the SD card is removed.

leonel85, 2022-12-27:
Why don’t you disable it during this operation?
M211 S0 ; Disable Software Endstop

mjaschen, 2022-12-27:
> Have you found the Special menu in firmware 1.4.3?

Yes, it was available via the print menu.

> Why don’t you disable it during this operation?

Two reasons:

1. because I didn't know until now that there's a GCODE command for that (and I didn't verify yet if that command changes the described behaviour, will try this tomorrow)
2. because it wasn't needed in the past and it would be extra work (open serial connection or OctoPrint, typing and sending command)

mjaschen, 2022-12-28:
(ps: `M211 S0` worked indeed, it's possible to move further down on the z-axis now; it would be nice if the z-endstop is automatically disabled when mesh-leveling is active)

anno0, 2023-01-01:
> 

With which tool can I create such a bed level graphic?

mjaschen, 2023-01-02:
> With which tool can I create such a bed level graphic?

That's from the “Bed Visualizer” plugin for OctoPrint.

DominicDevelop, 2023-01-07:
Sadly I got the same issue. Seems like using M211 S0 helps. Tho this can't be the way :(

anno0, 2023-01-07:
As workaround it helps to set the bed a little bit higher that the nozzle is a bit under zero point for Z.

DominicDevelop, 2023-01-07:
Do I have to reactivate z-endstop after mesh leveling or would the mesh not work then?

anno0, 2023-01-07:
In my case it hasn't worked. The Z endstop is the absolute minimum point. So you only can set a position for Z that is higher than the Z endstop. But I has not tested it with GCode M211 how this works.

DominicDevelop, 2023-01-07:
> In my case it hasn't worked. The Z endstop is the absolute minimum point. So you only can set a position for Z that is higher than the Z endstop. But I have not tested it with GCode M211 how this works.

I just sent the M211 S0 to my printer via Pronterface and could do the mesh leveling. But keep in mind to reenable it afterwards (read online to do so or you can destroy your bed with the next G28 command). Also when the printer is turned off and on again the M211 is enabled again (S1). So if you turned it off and want to level again now, disable M211 again with S0.

closed 2023-02-12 by github-actions[bot]