knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #413: Auto level X/Y not moving as before

SjoerdNLD, 2022-12-27:
### Bug Description

Auto level X/Y not moving as before

### Steps to Reproduce

Install newer 1.5.0 beta 2 FW

1. loaded the FW
2. load fw default
3. load M851 X-26.00 Y-29.00 Z-2.19
4. start auto level via special menu

**Expected behavior:** 
Moving along the builplate and recoding points.

**Actual behavior:** 
Moving only very short distance and making a loud noise

#### Additional Information

knutwurst, 2022-12-28:
Hi,
can you upload a video about the issue? When I tested it, it worked perfectly.

SjoerdNLD, 2022-12-29:
Hi,

Thx for your resonse, the video is attached.

1st is beta.2 
https://user-images.githubusercontent.com/46634319/209924884-b0b4d91a-872a-47c2-b467-58e24c396b4d.mp4
2nd is the previous beta which I loaded after + M851.
https://user-images.githubusercontent.com/46634319/209924907-e4628d0b-94db-43e0-a702-13da890ba811.mp4

SjoerdNLD, 2023-01-18:
Hi,

Any clue how I could fix this? Or how to start to find the problem?

Video when not working 1.5.0 beta 2 and 3
[not working 1.5.0 beta 2 and 3.webm](https://user-images.githubusercontent.com/46634319/213296335-e391d432-aa5b-455c-9fa9-e41b2354c7e1.webm)

Video when working 1.5.0 beta 1
[working 1.5.0 beta 1.webm](https://user-images.githubusercontent.com/46634319/213296379-8b4bcd39-bf37-46a8-9d43-6b9d7abd4ffa.webm)

rgds Sjoerd

SjoerdNLD, 2023-01-28:
> Hi, can you upload a video about the issue? When I tested it, it worked perfectly.

Any clue where to look?

closed 2023-03-05 by github-actions[bot]