knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #414: resume print

attilan6, 2022-12-28:
is there any way to;
- enabling power loss recovery feature.
- pause and resume print via touch screen for filament color changing.

may be giving up some other features for adding these two?

knutwurst, 2022-12-28:
Yes, but there is a lot to do to prepare the firmware for it. My plan is that the final 1.5.0 will not have any of these bugs. However, I don't know if power-loss-recovery will come back in, because the feature puts a lot of strain on the SD card.

HaGueRe, 2023-01-01:
In a German Speaking Forum, I wrote a little Workaround for this known issue with color change. It works fine with PrusaSlicer and also with modification of the color change postscript in Cura.
If you want to try it, have a look here and follow the instructions:
https://drucktipps3d.de/farbwechsel-ohne-m600-eine-moegliche-loesung/

closed 2023-02-07 by github-actions[bot]