knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #417: BLTouch fährt nicht aus

Viech83, 2023-01-05:
Hallo zusammen,

ich habe den I3-Mega mit Trigorilla 1.0 und die Firmware "MEGA_S_BLT_10_v1.5.0-b3" aufgespielt.
Der BLTouch liegt wie in der Anleitung beschrieben auf S0 (Der einzige Anschluss mit im originalzustand vorhandenen Kontakten).
Nach dem einschalten fährt dieser auch 2x Aus/Ein, die Verkabelung sollte also passen.
Die Kommunikation über Pronterface funktioniert auch (Extruder kalibriert).

Ich kann jedoch den BLTouch nicht ausfahren.
M280 P0 S10 erzeugt keine Reaktion oder Fehlermeldungen.

Über Hilfe wäre ich sehr dankbar !

Liebe Grüße,
Philipp

c-poxi, 2023-01-10:
Servus viech. . 
probier mal die finale 1.4.4 version - hab den bltouch auch grad installiert - auch umverkabelt auf pin 21 und dann sogar noch an neuen gekauft.
mit 1.4.4. gehts. (des hätt i vielleicht früher probieren sollen)

Viech83, 2023-01-20:
Servus c-poxi,

leider funktioniert es mit 1.4.4 auch nicht :-(
Ich kann den bltouch leider noch immer nicht ausfahren.

Was mich irritiert sind die unterschiedlichen Angaben der Servopins für das Trigorilla V1.0.
einerseits finde ich diese Grafik mit  
Servo 0 - Pin11
Servo 0 - Pin6
Servo 0 - Pin5
Servo 0 - Pin4
 [image]

andererseits steht im Beginners Guide:
Servo 0 - Pin7
Servo 0 - Pin6
Servo 0 - Pin5
Servo 0 - Pin4
https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/Beginner's-Guide-(Deutsch)#3-identifikation-des-mainboards

Leider kenne ich mich nicht damit aus, wie ich die Pins in der Firmware ändere und diese kompilieren kann.
Damit muss ich mich evtl. beschäftigen.

Aber die Belegung der Pins in der Firmware scheint nicht mit meinem Board übereinzustimmen.
Mein Board scheint aber das 1.0 (nur Logo ohne Versionsnummer) zu sein.

[image]

c-poxi, 2023-01-21:
servus viech. 
eins vorweg -bin selbst kein programmierer.(und hab auch den mega x - somit weiß ich nicht wie deine hotendplatine aussieht -)  dein board scheint auch speziell zu sein - lt. beschreibung eins von den ersten mit dennoch anderen anschlüssen - also ein 1.0.. würd mal die pins durchmessen wo plus und minus liegt . das andere sollte signal dann sein - auch aufpassen bei einer servoverlängerung.
wenn der bltouch ausfährt beim einschalten sollte die grundverkabelung passen . ist jetzt nur noch die frage wo du den z-min(weiß) angeschlossen hast. 
wennst nach anleitung vorgegangen bist sollte das gehen. wenn nicht dann müsstest das weiße kabel bei scl(=pin21) und das schwarze bei ground darüber einstecken ( sind die pins die etwas über der grünen platine im linken unteren eck liegen. das müsstest dann in der firmware kompilieren.
https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/Howto:-Firmware-mit-PlatformIO-selbst-kompilieren
bis punkt 7 durcharbeiten - dann sollts funktionieren. (hoffentlich)

p.s. hoffentlich ist das keine aktuelle aufnahme von deinem board - da seh ich nämliche keine kabel weggehen vom servoport.

closed 2023-02-26 by github-actions[bot]