knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #424: BL-Touch Problem / Autoleveling schlägt immer fehl.

Enie1, 2023-01-14:
Hallo zusammen,

ich habe mit meinem I3 Mega S und dem BL-Touch Sensor folgendes Problem:

Der BL-Touch Pin fährt zum Anfang 2 Mal zum Test ein und aus.
Die Kalibrierung habe ich gemäß der Anleitung vorgenommen und die richtige Firmware ([MEGA_S_TMC_BLT_10_v1.4.3.hex])installiert.
Mein Problem ist jedoch das sich zwar das erste Autoleveling starten lässt (Sowohl via Display, als auch mit G29), jedoch fährt der Pin nur beim ersten Punkt heraus.

Ab dem zweiten Messpunkt fährt der Sensorpin nicht mehr heraus und die Nozzle bohrt sich gnadenlos in mein Druckbett.

Ich habe nun bereits die Steckerbelegung mehrfach überprüft, obwohl ich überzeugt bin, dass hier kein Fehler vorhanden ist.

Entgeht mir hierbei irgend etwas?
Sollte mit G29 nicht ein erstes Mesh erstellt werden?
Liegt es an der Firmware, oder eher am Sensor?

Für eure Hilfe wäre ich sehr Dankbar.

MfG

v-a-d-e-r, 2023-01-23:
FW: MEGA_S_DGUS_TMC_BLT_10_v1.5.0-b3
Habe das gleiche Problem mit der FW 1.5.0b3. Fährt zur 1. Postion, dann zur 2. und Feierabend ist. Log:

//action:notification Settings Stored
echo:Invalid mesh.
Error:Failed to enable Bed Leveling
Error:Failed to enable Bed Leveling

echo:Bed Leveling OFF
echo:Fade Height OFF

v-a-d-e-r, 2023-01-23:
Update: Lag am BLTouch. Der Stift fällt nicht immer kpl. runter, wenn das Halte-Magnetfeld abgeschaltet wird. Er "klemmt" also irgendwo. Ok, dann habe den Stift mit Pinzette vorsichtig raus geholt bis zum Anschlag und mit Kältespray einfach runtergekühlt, damit er etwas schrumpft und nicht mehr klemmt. Dann sofort das Leveling gestartet und siehe da, er hat alle 25 Testpunke angefahren und gespeichert! Vielleicht hilf es ja jemandem... ;-)

closed 2023-02-28 by github-actions[bot]