knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #425: How to wire BLTouch on Trigorilla v.0.0.2?

DominicDevelop, 2023-01-15:
Hey guys :) So I bought a BLTouch for my Anycubic i3 Mega S and it has the Trigorilla v0.0.2 but the Guide has another Board version. In my case the pins are PS ; S1 ; S2 ; S3 ; S4 and not PS ; S0 ; S1 ; S2 ; S3 but is S1 still correct? Where can I find the manual for this board to check?

Hallo ihr Lieben :)
Ich hab mir den BLTouch für meinen Anycubic i3 Mega S gekauft und dieser hat das Trigorilla v0.0.2 aber der Guide ist leider für eine andere Version. In meinem Fall sind die Pins beschriftet mit PS ; S1 ; S2 ; S3 ; S4 und nicht PS ; S0 ; S1 ; S2 ; S3 aber ist der S1 trotzdem richtig? Wo kann ich die Anleitung zum Board finden für Informationen?

stklcode, 2023-01-16:
It's always the lowest numbered servo port, typically the only servo port with pin headers soldered by default. On the 0.0.2 board it's `S1`.

In the software, they have always been called `SERVO0` to `SERVO3`, no matter if the labeling is S1-S4 or S0-S3.

Board description: [Beginner's Guide (English)](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/Beginner's-Guide-(English)#trigorilla-002) / [Beginner's Guide (Deutsch)](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/Beginner's-Guide-(Deutsch)#trigorilla-002)

DominicDevelop, 2023-01-16:
> It's always the lowest numbered servo port, typically the only servo port with pin headers soldered by default. On the 0.0.2 board it's `S1`.
>
> In the software, they have always been called `SERVO0` [...]

Thanks for the answer! 😄 
Okay but is there some official or unofficial data sheet anywhere that I can use as reference for the future?

Btw. in the guide from @knutwurst there is this image where the cables are put through the existing hole that leads to the cable manager.
[image]
But in my printer (and probably in any other too) there is a small board that blocks the hole on the inside and I don't know how I am supposed to lead the cables through the hole since I couldn't find screws to un- and remount the board to put the cable through the hole. Any info on how to do it so that I don't need to drill a hole into my printer? 😅

stklcode, 2023-01-16:
> Okay but is there some official or unofficial data sheet anywhere that I can use as reference for the future?

This is not something board specific. The _BLTouch_ probe always uses a free servo port that just has to match the firmware configuration `Z_PROBE_SERVO_NR` (defaults to `0`) (https://marlinfw.org/docs/configuration/configuration.html#servo-z-probe)

There are some pinout descriptions of the board floating around the web but AFAIK nothing "official".

> But in my printer (and probably in any other too) there is a small board that blocks the hole on the inside [...] Any info on how to do it so that I don't need to drill a hole into my printer?

That board holds the green/red/black connectors, so it's always there.
There are 3 (?) hex screws from the inside. Little tricky to reach, but the connector board can be removed or at least moved away, so the plug fits.

Otherwise you might strip the 3-pin connector from the sensor and fiddle the 3 wires through one by one.

DominicDevelop, 2023-01-16:
> > Okay but is there some official or unofficial data sheet anywhere that I can use as reference for the future?
> 
> This is not something board specific. The _BLTouch_ probe always uses a free servo port that just has to match the firmware configuration `Z_PROBE_SERVO_NR` (defaults to `0`) (https://marlinfw.org/docs/configuration/configuration.html#servo-z-probe)
> 
> There are some pinout descriptions of the board floating around the web but AFAIK nothing "official". However, the board label shows "SERVOS" and the 3 lines are labeled +/-/S, so this part of the hardware is pretty much self-explaining.
> 
> > But in my printer (and probably in any other too) there is a small board that blocks the hole on the inside [...] Any info on how to do it so that I don't need to drill a hole into my printer?
> 
> That board holds the green/red/black connectors, so it's always there. There are 3 (?) hex screws from the inside. Little tricky to reach, but the connector board can be removed or at least moved away, so the plug fits.
> 
> Otherwise you might strip the 3-pin connector from the sensor and fiddle the 3 wires through one by one.

Thanks. That helped :)

yardi83, 2023-01-17:
Ich habe den BL Touch erst vor ein paar Tagen genau wie in der Anleitung (nicht die "alternative" Verkabelung) angebaut und habe auch ein Trigorilla 0.0.2. Hat auf Anhieb geklappt. Kannst dich also an die dort gezeigte Verkabelung halten.

Die Platine auf der die Stecker nach außen geführt werden kannst du abschrauben, dann die Kabel vom BL Touch durchziehen und wieder anschrauben. An der Seite wo die Kabel dann sind halt nicht so fest anziehen. Ich habe bei mir ein kleines Stück Plastik dazwischen geklemmt um die Kabel nicht einzuquetschen.

closed 2023-02-23 by github-actions[bot]