knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #427: [FR] TMC2208 v1.2 vref

leonel85, 2023-01-21:
### Description

I see the documentation on how so setup vref but my drivers has a potentiometer only in the bottom of the drivers. So I don’t understand how can I regolate Vref when it is mounted because in that case it’s not accessible. There’s a hole in the upper but if I rotate the screwdriver 🪛 the potentiometer in the bottom doesn’t move.
It’s uart labeled it’s means that it’s setup by the firmware or I have to do it somehow?
Thanks 

### Feature Workflow

1. Mount tmc2208 v1.2

#### Additional Information

* Provide pictures or links that demonstrate a similar feature or concept.

AlexanderBartash, 2023-01-24:
If you need to understand your options I recommend you watch this in the given order:

https://www.youtube.com/watch?v=Lx40lJkk9NQ
https://www.youtube.com/watch?v=Ut3Mj0AX5T4

https://www.youtube.com/watch?v=TubpTMFkbVs
https://www.youtube.com/watch?v=9V8iEEAB15g
https://www.youtube.com/watch?v=SyMNwmO5Nxo

https://www.youtube.com/watch?v=7VHwcEroHPk&list=PLGqRUdq5ULsOIIaBONPU65uH2s6iI6GqY&index=2

AlexanderBartash, 2023-01-24:
In short depending on the type of your board UART might require additional connections to be made, so this mode is not as simple as "remove the old driver, put a new one, enjoy". It is a bit more complex. Unless your old drivers already worked in UART mode and have the same pins.

leonel85, 2023-01-24:
Thanks for the support 
I use DuPont cables to power the drivers with the board and access to the potentiometer from bottom. I setup all and after that mount it in the board.
I really don’t understand why the potentiometer are on the bottom. In all videos it’s on the top

closed 2023-03-01 by github-actions[bot]