knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #437: [BUG] BLT on Pin21 with Mega does not work on 1.5.0-beta.3 but does not 1.4.4

danzbox, 2023-02-16:
### Bug Description

BLT on Pin21 with Mega does not work on 1.5.0-beta.3 but does not 1.4.4.

Changed z-probe min pin in ramps.h from Pin 2 to Pin21 (as shown in the great ["Firmware how-to"](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/Howto:-Compile-Firmware-with-PlatformIO)) before compiling the firmware and executing the also really good BLT guide the BLT does not execute Auto Leveling.

### Steps to Reproduce

1. Compile and Install Firmware 1.5.0 beta 3 with ramps.h modified to Pin 21 (z-probe min pin)
2. Follow BLT Guide to the end
3. Auto Level in the Special Menu

**Expected behavior:** BLT moves from Home to first position deploys Probe and executes first Position, then goes to the next and so on. It works for 1.4.4

**Actual behavior:** BLT moves from Home to first position deploys and retracts Probe several times and stops 2-3 cm above bed surface and makes an error noise.
On Pronterface the feedback is:
```
"Special Menu: Enter BLTouch Menu
Special Menu: BLTouch Leveling
Error:!! STOP called because of BLTouch error - restart with M999
Error:!! STOP called because of BLTouch error - restart with M999

Error:Printer stopped due to errors. Fix the error and use M999 to restart. (Temperature is reset. Set it after restarting)
Error:Printer stopped due to errors. Fix the error and use M999 to restart. (Temperature is reset. Set it after restarting)

Error:!! STOP called because of BLTouch error - restart with M999
Error:!! STOP called because of BLTouch error - restart with M999

Error:Probing Failed
Error:Probing Failed

echo:Settings Stored (738 bytes; crc 14960)
//action:notification Settings Stored
echo:Invalid mesh.
Error:Failed to enable Bed Leveling
Error:Failed to enable Bed Leveling

echo:Bed Leveling OFF
echo:Fade Height OFF
```

#### Additional Information
Auto level sequence starts 0:21 in this video
https://1drv.ms/v/s!AoG7FWrbzTX_4TPbtUH5c6h2oxB6?e=YTu1uD

xtraxion, 2023-02-17:
same behavior here.

BLTouch on Pin21.
Under 1.4.4 this works as desired.
With 1.5.0b3 I have the behavior described by danzbox.

Firmware was compiled several times as described in the wiki.
The 1.4.4 was also fresh compiled for the test with the same change as in 1.5.0b3.

Same result. 1.4.4 works, 1.5.0b3 does not.

Printer is a Mega X

Waldwegheizer, 2023-02-19:
Hi,genau das selbe Problem besteht bei mir auch. 
I3 mega s

Jemand schon eine Lösung?

-------------

Hi, I have exactly the same problem.
I3 mega s

Anyone already a solution?

eugenmeltis3pg, 2023-03-11:
Confirming this behaviour as well

closed 2023-04-16 by github-actions[bot]

hoovercl, 2023-09-04:
Was there any resolution to this?  I am experiencing the same behavior with all of the 1.5.x versions and attempting to use PIN 21 on a Mega S (I have not tried it on a Mega X, yet).

Version 1.4.4 with Z_MIN_PROBE_PIN set to 21 works just fine. I have built versions 1.5.0, 1.5.1 and 1.52 with the same setting and with all three versions, when I attempt a G29, I have the same result.

[image]

I did notice one slight difference in the movement of the carriage with the 1.5.x versions compared to 1.4.4  When a G29 is sent with 1.4.4, the carriage raises, the bed moves forward, the BLTouch probe pin is deployed and then it begins probing the bed.  With the 1.5.x versions, after sending the G29, the carriage raises, and as the bed moves forward, the carriage also attempts to move to the left (doesn't occur with V1.4.4).  Afterwards, the BLTouch pin extends, then retracts, extends, retracts, and then extends a third time at which point the error message is received.

As mentioned, the error does not occur with V1.4.4.  I have confirmed this using the build I made in October 2022, and a fresh build that was created from the original download and again changing the Z_MIN_PROBE_PIN changes to 21 in pins_RAMPS.h.  The size of the .hex file from October 2022, and the one just built are exactly the same size.  

I have seen it mentioned elsewhere that this stop error is due to incorrect connection of the BLTouch to the board, however that does not seem to be the issue here since the V1.4.4 builds are working correctly.

Using the <Start Auto Leveling> from the special menu, has similar results, the only difference is the behavior of the probe which will extend, retract, extend, retract, extend, short pause, retract, extend, retract, extent, retract, followed by some tones, and then it appears the printer resets.