knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #438: [FR] Support for Slice Thermistor

geezy54, 2023-02-16:
-->

### Changing out Stock hotend for bondtech hotend, with bl touch, and new thermistor and heater.

### Feature Workflow

1. I am doing a project that uses a bond tech LGX Shortcut Mosquito that I will be hooking up to my anycubic mega x, this is my first type of project like this. It also is going to have a BL touch mounted on it. I was so impressed by the firmware I found on your website that I decided to upgrade the printer which is proving to be very expensive.
2. I think most of the parts are just plug and play but the thermistor on my other hotend does not fit the new one. I was looking into it and you have to change the firmware for a new thermistor.
3. I did buy a new thermistor, but I have no clue how to change the firmware for printer. Should I just use the old one with an adapter or try to use the new one? I bought the slice one that is listed below in the links.
4. I will take any advice I can get on the subject, I have not started working on the project yet, but if anyone is willing to help point me in the right direction I would be grateful.
5. I have not opened it up yet, if anyone is willing to help I will open it and see which model the motherboard is.

Thank you,

-Greg

#### Additional Information
https://www.bondtech.se/product/lgx-shortcut-mosquito-printhead/
https://www.bondtech.se/product/lgx-shortcut-accessories-set-for-anycubic/
https://www.bondtech.se/product/slice-300c-thermistor-2m-bare-leads/
https://www.bondtech.se/product/mosquito-12v-50w-heater/

geezy54, 2023-02-16:
It currently has Mega_X_v.1.4.4.hex loaded if that helps.

geezy54, 2023-02-16:
[image]

MGThePro, 2023-02-20:
I'm no expert but afaik you'd only have to change the thermister type [here](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/blob/master/Marlin/Configuration.h#L658) based on [this](https://support.sliceengineering.com/portal/en/kb/articles/how-do-i-update-my-firmware-to-be-compatible-with-the-high-temperature-thermistor) information, then rebuild the firmware.

So in the case of the 300°C thermistor that you linked, the TEMP_SENSOR needs to be set to 5

Just make sure to build the firmware with the configuration you need (in your case the version for the Mega X with BLTouch support) in PlatformIO

geezy54, 2023-02-20:
Hello MG,
I appreciate the help. Thank you for pointing out what to change.

Where can I get the MEGA X with BLTouch Support source files for compiling?

I see the source files, and some of the files look like for Any cubic, but I am not sure how to find the correct source files for the MEGA X with BL TOUCH support.

Is there a way to just take the completed BL Touch Config and edit it? or that is not possible?

Thank you,

-Greg

MGThePro, 2023-02-20:
The source files are identical for every printer and modification, during the build process you specify your configuration. The build process is described [here](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/Howto:-Compile-Firmware-with-PlatformIO).

Notice how in the first image of section 5 there is a list on the left of printer configurations. In the picture, the configuration "MEGA_S_DGUS_TMC" is selected. Make sure you select the correct configuration for your printer. You should also read section 1 to 4 of [this](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/Beginner's-Guide-(English)) page to know what configuration you need for your printer.

I'm not entirely sure what you mean by editing the completed BLTouch config, but as long as you follow the wiring diagram from the wiki and using the correct BLTouch build configuration, you should only have to configure the position of the BLTouch as well as the Z offset, which can all be done via gcode rather than firmware modifications. However this is already beyond my scope of knowledge, I haven't done anything with a BLTouch yet.

geezy54, 2023-02-20:
Mg,
I think this is what I needed, the information on how to do it.
Thank you so much for your input, I will read this and compile the right files for my printer.

Have a great day, I will let you know how it goes.

closed 2023-03-28 by github-actions[bot]