knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #439: [BUG] busy protocol broken

repetier, 2023-02-22:
### Bug Description

I lately got complaints about timeouts in repetier-server. Testing 1.4.4 and 1.5.0 on anycubic i3 i found out that your busy messages do not appear when moves are causing a delayed "ok". So thought it is a marlin bug and insalled latest bugfix of marlin but that version behaves correct. So either you broke someting or they fixed it upstream. busy is enabled and with M400 or during G28 it appears, but not if moves cause blocking which is the most important part,

### Steps to Reproduce

Set host timeout to 3 seconds and print this file with logging enabled.
[busytest.gcode.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/10802864/busytest.gcode.zip)

**Expected behavior:** Print without timeouts

**Actual behavior:** Lots of timeouts like in this snippet:
```
Send:8:17:24.020: N45 G1 F1200
Recv:8:17:24.024: ok
Send:8:17:24.024: N46 G1 X100
Recv:8:17:24.027: ok
Send:8:17:24.027: N47 G1 X0
Recv:8:17:24.030: ok
Send:8:17:24.030: N48 G1 X100
Recv:8:17:24.033: ok
Send:8:17:24.033: N49 G1 X0
Recv:8:17:24.037: ok
Send:8:17:24.037: N50 G1 X100
Recv:8:17:24.851: T:21.91 /0.00 B:22.66 /0.00 @:0 B@:0
Recv:8:17:25.856: T:22.07 /0.00 B:22.81 /0.00 @:0 B@:0
Recv:8:17:26.861: T:22.03 /0.00 B:22.89 /0.00 @:0 B@:0
Recv:8:17:27.866: T:22.07 /0.00 B:22.77 /0.00 @:0 B@:0
Mesg:8:17:28.044: Warning: Communication timeout - resetting communication buffer.
Mesg:8:17:28.044: This means that a expected firmware response was not seen within the expected time.
Mesg:8:17:28.044: The typical reason is a communication error and print should continue after the communication reset.
Mesg:8:17:28.044: Connection status: Buffered:96, Manual Commands: 1, Job Commands: 25
Mesg:8:17:28.044: Buffer used:96 Enforced free byte:18 lines stored:11
Send:8:17:28.044: M117 ETE 00:03:15
Send:8:17:28.044: N51 G1 X0
Send:8:17:28.045: N52 G1 X100
Send:8:17:28.045: N53 G1 X0
Send:8:17:28.045: N54 G1 X100
Send:8:17:28.045: N55 G1 X0
Send:8:17:28.046: N56 G1 X100
Send:8:17:28.046: N57 G1 X0
Send:8:17:28.046: N58 G1 X100
Send:8:17:28.047: N59 G1 X0
Send:8:17:28.047: N60 G1 X100
Recv:8:17:28.871: T:22.11 /0.00 B:22.77 /0.00 @:0 B@:0
Recv:8:17:28.942: ok
Send:8:17:28.942: N61 G1 X0
Send:8:17:28.943: N62 G1 X100
Recv:8:17:29.875: T:22.15 /0.00 B:22.73 /0.00 @:0 B@:0
Recv:8:17:30.880: T:21.95 /0.00 B:22.66 /0.00 @:0 B@:0
Recv:8:17:31.884: T:21.95 /0.00 B:22.70 /0.00 @:0 B@:0
Recv:8:17:32.890: T:22.03 /0.00 B:22.62 /0.00 @:0 B@:0
Mesg:8:17:32.948: Warning: Communication timeout - resetting communication buffer.
Mesg:8:17:32.948: This means that a expected firmware response was not seen within the expected time.
Mesg:8:17:32.948: The typical reason is a communication error and print should continue after the communication reset.
Mesg:8:17:32.948: Connection status: Buffered:102, Manual Commands: 0, Job Commands: 13
Mesg:8:17:32.949: Buffer used:102 Enforced free byte:13 lines stored:12
Send:8:17:32.949: N63 G1 X0
Send:8:17:32.949: N64 G1 X100
Send:8:17:32.949: N65 G1 X0
Send:8:17:32.950: N66 M118 You should have seen some busy by now
Recv:8:17:33.895: T:22.03 /0.00 B:22.62 /0.00 @:0 B@:0
Recv:8:17:33.967: echo:busy: processing
```
you see line 50 was send and server waits for ok and gets no busy so assumes missed ok and sends more commands. This can cause fatal errors if it happens too often.

knutwurst, 2023-04-18:
I'll take a look at it. Thanks!

knutwurst, 2023-05-28:
When I put my printer in a busy state, I get a message every second. However, I have now changed a lot again and released a new 1.5.0 Beta 5. Maybe you can test it again with your setup?

closed 2024-05-05 by github-actions[bot]