knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #447: No Access to SD Card

ASTuser1, 2023-03-31:
I have a Mega M with a Trigorilla V1.1 Board (+ Blue/White Display) and i have problems to access the SD-Card with Firmware V 1.5.0-b3. As long the SD card is removed, i can access the Setup-Menu, but when the SD-Card is inserted, i see the additional "\"-Folder, but the Machine doen´t respond any longer to any input=hangs. Sending M20 via Pronterface gives me a File list, M20 tells me "SD Card ok".

What could be wrong?

ASTuser1, 2023-03-31:
Edit: Doesn´t happen on 1.4.4.

ASTuser1, 2023-03-31:
Also having trouble with Filament change on 1.4.4 during SD-Card-Print. Nozzle parks, releases the filament, Status on LCD goes to "paused"  after pressing "continue" the filament is fed in as supposed, but after that the printer stalls. Sending M118/M24 or whatever via Cura wont resume the print. the same happens during printing directly via USB. I suppose the buffer is being filled up and no other external command is accepted.

ASTuser1, 2023-03-31:
Ok, the Filament change won´t resume because of this:

`echo:busy: processing
//action:paused
//action:prompt_end
//action:prompt_begin Pause
//action:prompt_button Dismiss
//action:prompt_show
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:Insert filament and send M108
//action:prompt_end
//action:prompt_begin Nozzle Parked
//action:prompt_button Continue
//action:prompt_show
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
//action:prompt_end
//action:prompt_begin Paused
//action:prompt_button PurgeMore
//action:prompt_button Continue
//action:prompt_show
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user`

Because the display cannot show "Purge more" and "Continue", it stalls in a "pause for user-loop", which i cannot break by sending various commands via serial. It wants TFT UI input to continue, no matter what command comes in via serial.

ASTuser1, 2023-04-01:
SD works. The cause was that the SD-Card contained two Directories (Anycubic factory written with spaces and some Chinese Characters), which presumably cannot be handled by either Marlin and Long File Format (since 8:3 though M20 worked)- or the Display got a Char which it did misunderstood and locked up. For whatever reason.

ASTuser1, 2023-04-01:
Got Filemant Change also working. After issuing M108 and Fillament being fed in, i also have to issue also M876 S1 through the Emergency Parser to overcome the "purge More" and "Continue"-Message, which are not shown on Display

knutwurst, 2023-04-18:
Hi, sorry for the late reply!
The information is very valuable and thank you so much for sharing it. :)

I'll take a look!

Edo681, 2023-05-02:
Hallo Ich habe ich habe Anycubic Mega S habe ein Problem wen ich beim Drucken Pause setze kann ich nicht mehr fortsetzen mit Version 1.1.9 ging gut mit 2.0 nicht mehr

knutwurst, 2023-05-24:
Already fixed in https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/commit/4d6715761d77343408a0ec95c31f488d8cfad035.

You can use 1.5.0 Beta 4 to get everything you need :)

ASTuser1, 2023-06-15:
Sorr for the late replay, thanks for the Fix, i will try it as soon as i can (currently the Printer is beinge heavily used). About the Filament issue - any chance to add M876 S1 after the Filament is being fed in or modify pause.cpp, Line 283 ff?

`      #if M600_PURGE_MORE_RESUMABLE
        if (show_lcd) {
          // Show "Purge More" / "Resume" menu and wait for reply
          KEEPALIVE_STATE(PAUSED_FOR_USER);
          wait_for_user = false;
          #if EITHER(HAS_MARLINUI_MENU, DWIN_LCD_PROUI)
            ui.pause_show_message(PAUSE_MESSAGE_OPTION); // Also sets PAUSE_RESPONSE_WAIT_FOR
          #else
            pause_menu_response = PAUSE_RESPONSE_WAIT_FOR;
            //pause_menu_response = PAUSE_RESPONSE_RESUME_PRINT; // KNUTWURST
          #endif
          while (pause_menu_response == PAUSE_RESPONSE_WAIT_FOR) idle_no_sleep();
        }
      #endif`