knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #449: [BUG] Stop success endless loop

anno0, 2023-04-01:
### Bug Description

Sometimes I have to stop a print before it really has started. In the eat up or cool down phase I stop the print. A white dialog "stop successful" is displayed. Often I can not close this dialog correctly.

### Steps to Reproduce

1. Start a print
2. Wait for a short time
3. Stop the print
4. Try to close the dialog "stop successful" while pressing the OK button 
5. After pressing the OK button the dialog is displayed again and again

**Expected behavior:**
If the button is pressed once the dialog is closed and will not display again after the print is started and stopped again.

**Actual behavior:** 
Sometimes there is a endless loop which displays the dialog again.

#### Additional Information
Anycubic MEGA X with DGUS clone display.
Release 1.4.4

knutwurst, 2023-04-18:
Can you see if the problem still exists in 1.5.0?

anno0, 2023-04-21:
I have to check this first this weekend

anno0, 2023-04-23:
This bug still exists in 1.5.0 Beta 3.
I updated the steps to reproduce and attached a video of the main menu.

[https://www.dropbox.com/s/58k5vkic8x53cl6/VID_20230423_204634.mp4?dl=0](url)

**Steps to Reproduce**

1. Start a print
2. Wait for a short time
3. Stop the print
4. Try to close the dialog "stop successful" while pressing the OK button (this works in the print menu)
5. **Go to the main menu where the print button is displayed. Here the "print successful" dialog is displayed in a endless loop.** 
6. After pressing the OK button the dialog is displayed again and again

knutwurst, 2023-05-24:
Fixed in https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/commit/52dc811774a5d1cb2347e5fb6c13e26cb94623ad

Use 1.5.0 Beta 4 to get everything you need :)

closed 2023-05-24 by knutwurst