knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #450: [BUG] (short description)

KnoXe1993, 2023-04-02:
### Bug Description
Setup:
MEGA_S_TMC_BLT_10_v1.5.0-b3.hex
tmc2208 fysetc
Anycubic Mega S Generation 1
3D Touch V3.0 
Cura Slicer

**Die Nozzel fährt nicht hoch bzw. die Z Achse bewegt sich nach dem ersten Layer nicht nach oben. 
Unter 1.4.4 funktioniert alles ganz großartig.** 

Super Arbeit!

knutwurst, 2023-04-18:
Kannst du bitte genau beschreiben, was das Problem ist? Was heißt, dass die Düse nicht nach oben bewegt? Gar nicht mehr? Oder nur unzureichend? Was funktioniert sonst? Hast du den Z-Offset korrekt eingestellt?

KnoXe1993, 2023-04-19:
Moin Oliver,

das Problem hat sich erledigt. Cura hat über die Funktion „Extruder-Versatzwerte auf GCode anwenden“ den Z-Offset überschrieben und nach entfernen der Berechtigung läuft alles. Leider habe ich mit verschiedenen Cura Versionen gearbeitet und in nicht allen war der Haken gesetzt, was zur Fehlannahme führte, dass die Firmware dafür verantwortlich wäre.

Du machst da richtig gute Arbeit, den Café hast du dir mehr als verdient 😉

closed 2023-05-24 by knutwurst