knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #451: Filament Change leads to USB-Reset [BUG] (short description)

ElPainis, 2023-04-15:
### Bug Description

My printer loses Connection to my device when I execute the filament-option on the build-in Screen. One time the device resets itself to firmware-default what leads to even more issues.

### Steps to Reproduce

1. Plug in a device (Octopi for example) in the USB-Port.
2. Connect that device with printer
3. Chose Menü-Point filament
4. Choose Load or Unload and wait for heatup
5. The Feeder Starts to Unload and the issue appears after a few cycles.

**Expected behavior:**
Printer Unload/Loads filament and usb-device has Connection.

**Actual behavior:**
Printer loses Connection to USB device and repoduces even more issues.

#### Additional Information

knutwurst, 2023-04-18:
Which firmware version are you using?

ElPainis, 2023-04-18:
Im using MEGA_P_DGUS_BLT_10_v1.4.4.hex

*GERMAN*
Ich benutze die MEGA_P_DGUS_BLT_10_v1.4.4.hex Firmware.

knutwurst, 2023-04-18:
Have you tested whether the problem still occurs with 1.5.0?

ElPainis, 2023-04-18:
No. Is there a stable 1.5.0?

knutwurst, 2023-04-18:
No, this is still a beta. But it would be great if you could test this one, because the code base is a completely new one.

ElPainis, 2023-04-18:
I will try. Can I ask you an diffrent question not related to the topic but to the firmware?

knutwurst, 2023-05-24:
Fixed in https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/commit/52dc811774a5d1cb2347e5fb6c13e26cb94623ad

Use 1.5.0 Beta 4 to get everything you need :)

closed 2023-05-24 by knutwurst