knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #452: [FR] difference between this FW and stock marlin

Ingegneus, 2023-04-18:
it's not really a feature request, but more an inquiry. what are the differences between this version of marlin and the stock version? I read the faq, but there isn't all that much info on this. maybe my request is to expand the faq to address this

knutwurst, 2023-04-18:
This is a hybrid firmware that does not require configuration files for individual printers. And the advantage over Stock Marlin should be the full support of the display and the additional functions in the special menu, which are not otherwise available.

However, some guys already pushed some features into the Marlin Upstream and therefore some features such as FlowControl are also available in Stock Marlin.

The advantage is perhaps also that all supported printers are preset without further configuration.

It's an it-just-works solution... at least it was when I built the firmware back then. And back then the Anycubic Display was not supported at all and even longer the DGUS Clone Display was not supported either. If that's the case in Marlin now, it's because someone copied the features from me. But that's okay too. It's open source.

Ingegneus, 2023-04-20:
> This is a hybrid firmware that does not require configuration files for individual printers

what does hybrid mean in this context? and it doesn't need config files, because there is one "master" file that contains all settings that are selected via the build flags right? so I could "extract" the config for the mega x and use it more or less as is (removing all the ENABLE statements) for marlin right?

ok I understand thank you very much!

knutwurst, 2023-05-24:
@schmttc I'm actually working on making my code as modular as possible and implementing a lot via ExtUI lately. So yes, some things should also flow directly into the current marlin sources in the near future. I just need to fix some bugs :)

closed 2023-05-24 by knutwurst