knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #453: [BUG] list items missing in the print list

Ingegneus, 2023-04-20:
### Bug Description

I haven't figured out yet what triggers this, but very often some items in the print list are not show. when I tap the empty space the text is shown in red (as in "selected")

### Steps to Reproduce

1. upload a file to sd
2. open the print menu

**Expected behavior:** all list items present

**Actual behavior:** not all list items present

#### Additional Information

Before I tap anything:
[image]

after I tap the missing item. also notice how the folder as a .gcode appended where it shouldn't be.
[image]

I might have to add that while flashing the FW I always get "avrdude: verification error, first mismatch at byte 0x8e6c 0xbf != 0xff" , but everything seems to be working fine

knutwurst, 2023-04-23:
Which firmware version do you use? 1.4 or 1.5?

knutwurst, 2023-04-26:
Fixed in current master branch.

commit 4d671576 (2023-04-23): Revert c2c950b and 41c1600 and put changes into 36c8aee to recreate the 1.4.x filane behaviour. This should fix #453 and #455. It's untested atm and can include bugs.