knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #455: [BUG] (short description)

anno0, 2023-04-23:
### Bug Description
Display error of hidden folder. I have seen this also in another bug https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/issues/453

### Steps to Reproduce

1. Create a hidden folder on a SD card. This is normally a trash folder in Ubuntu. 
2. Insert the SD card in the printer 
3. Navigate to the print menu 

**Expected behavior:** This folder should not be shown in the print menu and even should not have a gcode extension.

**Actual behavior:** The gcode extension is display at position of the buttons  

[image]

#### Additional Information
Firmware 1.5.0 beta 3

knutwurst, 2023-04-23:
Can you compile the firmware by yourself? The issue *should* be fixed now, but is untested atm. I'll try it out if I have time and build a new Beta, but for now, the current master should work fine.

anno0, 2023-04-23:
I do not have compiled this project yet myself but I will figure it out next week.

anno0, 2023-04-25:
I compiled the master branch and flashed my printer. The print menu entries with gcode file extension look good again.

knutwurst, 2023-04-26:
>I compiled the master branch and flashed my printer. The print menu entries with gcode file extension look good again.

That's great to hear. Thank you for testing!

closed 2023-05-27 by knutwurst