knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #456: [BUG] Stop sequence

anno0, 2023-04-25:
What about this reported bug? 

I have not found the code in the master which was introduced here https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/issues/388 

> I changed the following in configuration_adv.h:
#define EVENT_GCODE_SD_ABORT "G92 E0\nG1 E-5 F1000\nG28X\nG1 Y175 F3000"
But it is limited what can be entered here. When it is too much, the firmware can't be flashed to the printer anymore. I had this issue yesterday. The following example is not working:
EVENT_GCODE_SD_ABORT "G92 E0\nG1 E-5 F1000\nG28X\nG1 Y175 F3000\nM104 S0\nM140 S0\nM107\nM84"

knutwurst, 2023-05-24:
Yes, you can only use some small commands, but maybe it's suitable for you when you use the park function (see: https://marlinfw.org/docs/gcode/G027.html) and configure your parking parameters.

knutwurst, 2023-05-28:
Is this still an issue?

anno0, 2023-05-29:
I will try it make a branch and compile this version with another stop Sequence. I think it is a good solution. So this bug can be closed

closed 2023-05-30 by knutwurst