knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #457: [FR] BL Touch Leveling Y offset

geezy54, 2023-05-04:
Need help please.

I installed my BL touch and the Y offset I set to -61.00mm.

It now hits the probe on the plate but I do not know how to perform the BL touch set up because when I try to go through the guide then it just goes off the front side of the build plate.

If anyone could please steer me in the right direction I would be grateful. Thank you.

Here are some pictures:
Here is where the Y offset lands - is this far enough in?
[image]

When I start doing the guide it just drives it right off the side, Any ideas on what I am doing wrong?
[image]

geezy54, 2023-05-24:
1. Heat up the bed.
2. Send command: G28
3. Send command: G90
4. Send command: G1 Z10
5. Send command: G1 X40 Y101 F4000 (Changed to ensure the sensor stays on the bed)
6. Send command: M280 P0 S10
7. Change to relative positioning: G91
8. Slowly lower the nozzle with G1 Z-1 or G1 Z-0.1 or G1 Z-0.02 until the sensor is triggered. NOT FURTHER!
9. If the sensor has triggered and flashes: M280 P0 S160
10. Send command: M114 (This gives the current position of the nozzle, e.g., Recv: X:40.00 Y:101.00 Z:1.30 E:0.00 Count X:3200 Y:8080 Z:520)
11. Change to absolute positioning: G90
12. Send command: G1 X40 Y40 F4000 (Since your X offset is 0, no change is needed in X. But for Y, you still need to adjust for the sensor's offset, so Y=101+(-61))
13. Change to relative positioning: G91
14. Place a sheet of paper under the nozzle
15. Slowly lower the nozzle with G1 Z-1 or G1 Z-0.1 or G1 Z-0.02 until the paper is difficult to move.
16. Send command: M114 (This gives the current position of the nozzle, e.g., Recv: X:40.00 Y:40.00 Z:0.70 E:0.00 Count X:3200 Y:3200 Z:280)
17. Subtract the Z value from point 16 from that from point 10 -> 1.30 - 0.70 = 0.6 mm -> This is our Z offset.
18. Send command: M851 Z-0.6 (be aware of the negative sign!)
19. Save settings with: M500
20. Carry out Autoleveling again via the display. Finished.

Can anyone just look this over to see if I made any mistakes please?

knutwurst, 2023-06-01:
This isn't rocket science, just a guide to touching the exact same position on the bed with the nozzle AND the sensor to find out the Z offset. Nothing more.

At point 5 you have to find a position (just drive there manually) where the sensor can touch the bed. X40 Y101 is of course nonsense in your case because your sensor is much further at the front.

Up to point 9 it is only tested whether the sensor is working correctly.

At point 10 you read out the current position of the sensor. And as a return value you get the X and Y value of the position. You just have to remember it so that you can then move the nozzle to exactly the same position. It's good for nothing else :)

At point 12 you then move the nozzle(!) to this position, which the sensor just had.

...And then gradually lower the nozzle until it touches the bed. And if that happens, you send M114 again to read out X, Y and also Z values. This time, however, X and Y is unimportant, you want to know how far down you had to go before the nozzle also touches the bed. You already know the value of the sensor. Z is important here.

And in the end you just subtract the values and you have the distance between sensor and nozzle.

That's all.

Alternatively, you can simply estimate the Z value, start a print and gradually change the Z offset. I just think that this takes longer than measuring or calculating it once.

geezy54, 2023-06-01:
Thank you for this, I really appreciate you going over it for me.

The printer has had a little mishap I think the thermistor cable was damaged a little and now it's not functioning well.

I did not bundle it tight enough so it was my fault.

But no worries I will replace it over the weekend and check this out.

Thanks again, and have a nice rest of your day.