knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #458: [FR] AI3MP Autoleveling in 0.01 Schritten

ElPainis, 2023-05-05:
### Description

In der P-Version vom I3 Mega sind TMC2208-Treiber in der Z-Achse verbaut. Trotzdem bewegen sich die Stellmotoren über Terminal-Eingaben und über Autoleveling nur in 0.02-Schritten.

### Feature Workflow
Ein ansteuern der Stellmotoren in 0.01-Schritten im Auto-Leveling-Menu.

knutwurst, 2023-05-24:
Das ist technisch gar nicht möglich, da die Motoren sich nur in 1,8° Schritten bewegen können.

Siehe: https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/FAQ-(deutsch)#wieso-sehen-drucke-mit-01mm-layerhöhe-oft-schlechter-und-unsauberer-aus-als-bei-02mm

closed 2023-05-24 by knutwurst

knutwurst, 2023-05-24:
Nein, können sie nicht. TMC2208/2209 lösen zwar feiner auf, können aber trotzdem nur mit ganzen Schritten angesteuert werden. Es ist nur der Übergang zwischen den Schritten weicher.

Mega M, S und P können nur 0,02mm fahren.

marvinlehmann, 2023-07-28:
> Nein, können sie nicht. TMC2208/2209 lösen zwar feiner auf, können aber trotzdem nur mit ganzen Schritten angesteuert werden. Es ist nur der Übergang zwischen den Schritten weicher.
> 
> Mega M, S und P können nur 0,02mm fahren.

Moin. Ich hätte diesbezüglich dann auch mal ne Frage zum Mesh Leveling. Im Menü habe ich da ja z.B. `<Z Down 0.1>` - würde in diesem Fall dadurch nicht ne Ungenauigkeit reinkommen (?). Wäre 0.08mm oder so nicht "besser"?