knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #460: [BUG] BLTouch Probe Fails to Redeploy During Auto-Leveling on Anycubic i3 Mega

faultyDaantje, 2023-05-24:
### Bug Description

During the auto-leveling process on the Anycubic i3 Mega 3D printer, the probe of my 3D-touch (BLtouch clone) deploys and triggers succesfully for the first (and sometimes second) point, after the probe triggers, it rectract and doesn't redeploy for the next point. 

[Video of the problem](https://youtu.be/Cpll98PM5qA)

I have had the probe work on the first point every time which makes me think the problem is not in my 3d-touch or in the cable.
I have calibrated the Z-offset many times and have tried many diffrent offsets, but none worked.

I have also tried diffrent versions of this firware, I have the anycubic i3 mega with the 1.1 motherboard, so I first tried the [MEGA_BLT_11_v1.4.4.hex](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/releases/download/1.4.4/MEGA_BLT_11_v1.4.4.hex) and after that I tried the [MEGA_BLT_11_v1.5.0-b4.hex](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/releases/download/1.5.0-beta.4/MEGA_BLT_11_v1.5.0-b4.hex) version, both have the same problem.

I couldn't find this problem anywhere on the internet so im curious if I'm just stupid or there is another problem.

Maybe the terminal output before the probing fails has something to do with the problem?
terminal output:
```
Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
Send: M113 S2
Recv:  T:199.89 /200.00 B:60.00 /60.00 @:22 B@:32
Recv:  T:199.97 /200.00 B:59.99 /60.00 @:22 B@:34
Recv:  T:200.03 /200.00 B:59.99 /60.00 @:23 B@:33
Recv: echo:busy: processing
Recv:  T:199.97 /200.00 B:59.99 /60.00 @:24 B@:33
Recv:  T:200.03 /200.00 B:59.99 /60.00 @:23 B@:33
Recv: X:2.00 Y:2.00 Z:0.00 E:0.00 Count X:160 Y:160 Z:0
Recv: ok
Recv:  T:200.00 /200.00 B:60.01 /60.00 @:24 B@:29
Recv:  T:200.00 /200.00 B:59.97 /60.00 @:24 B@:35
Recv:  T:199.83 /200.00 B:59.99 /60.00 @:26 B@:33
Recv: echo:busy: processing
Recv:  T:199.83 /200.00 B:60.01 /60.00 @:26 B@:29
Recv:  T:199.83 /200.00 B:60.01 /60.00 @:26 B@:29
Recv: echo:busy: processing
Recv:  T:199.83 /200.00 B:59.99 /60.00 @:25 B@:33
Recv:  T:199.97 /200.00 B:60.00 /60.00 @:24 B@:30
Recv:  T:200.00 /200.00 B:60.01 /60.00 @:24 B@:28
Recv: echo:busy: processing
Recv:  T:200.10 /200.00 B:59.97 /60.00 @:23 B@:35
Recv:  T:200.17 /200.00 B:60.06 /60.00 @:23 B@:20
Recv: //action:notification Probing Failed
Recv: Error:Probing Failed
Changing monitoring state from "Operational" to "Error"
Send: M112
Send: N2 M112*35
Send: N3 M104 T0 S0*34
Send: N4 M140 S0*97
Changing monitoring state from "Error" to "Offline after error"
Connection closed, closing down monitor
```

### Steps to Reproduce

1. Install the MEGA_BLT_11_v1.4.4.hex or the MEGA_BLT_11_v1.5.0-b4.hex firmware on an anycubic i3 mega with the 1.1 motherboard.
2. Start probing with a BLtouch clone
3. watch it fail

**Expected behavior:** 
The probe redeploys after triggering and the autolevel doesn't fail

**Actual behavior:** 
The probe doesn't redeploy and the autoleveling fails.

#### Additional Information
[Video]([MEGA_BLT_11_v1.5.0-b4.hex](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/releases/download/1.5.0-beta.4/MEGA_BLT_11_v1.5.0-b4.hex))
- The printer is connected to a raspberry pi zero running octoprint
- The sensor is a cheap clone of the BLtouch from aliexpress, the 3dtouch
- I don't know if this even is a bug, but this seemed like the best place to ask my questions as the facebook group is only in German.

knutwurst, 2023-05-24:
This is not a firmware bug, but in most cases either bad wiring causing the error or an incorrectly set Z offset.
See: https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/FAQ-(english)

faultyDaantje, 2023-05-25:
Hey, thank you for your comment, but I haven't found a solution yet, I have checked all the wiring and all looked good, also if the wiring would be bad the 3D-touch would be less consistent, but the probe only doesn't deploy when on the second or third point of autolevel, the probe always deploys on the first point and also works when manually testing the probe with gcode commands like M280 P0 S10 and M280 P0 S160.
I also don't think the Z offset is incorrect, I have used your guide on finding the offset multpiple times and everytime I endud up with a Z-offset of -4,00 mm.
I have tried some other values like +4,00 just to be sure, but i'm pretty sure the Z-offset should be -4,00.

Maybe I'll try looking with the oscilloscope when i have the time for it.

Do you have any other ideas on why it might not be working?

knutwurst, 2023-05-25:
What exactly does the console say about e.g. Pronterface when the error occurs?

faultyDaantje, 2023-05-25:
This happens in my pronterface console when the error occurs:

```
RECV: echo:busy: processing
RECV: echo:busy: processing
RECV: echo:busy: processing
RECV: X:2.00 Y:2.00 Z:0.00 E:0.00 Count X:160 Y:160 Z:0
RECV: ok
RECV: echo:busy: processing
RECV: echo:busy: processing
SENT: M105
RECV: echo:busy: processing
RECV: Error:Probing Failed
Error:Probing Failed

RECV: X:40.33 Y:44.70 Z:24.00 E:0.00 Count X:3226 Y:3576 Z:9600
RECV: ok
RECV: echo:Settings Stored (738 bytes; crc 58288)
RECV: //action:notification Settings Stored
RECV: ok
RECV: ok
RECV: ok
RECV: ok
RECV: ok
RECV: ok
RECV: ok
RECV: ok
RECV: ok
RECV: echo:Invalid mesh.
RECV: Error:Failed to enable Bed Leveling
Error:Failed to enable Bed Leveling

RECV: echo:Bed Leveling OFF
RECV: echo:Fade Height OFF
RECV: ok
```

faultyDaantje, 2023-05-28:
Hey, I ended up using [Marlin-AI3M-BLTouch](https://github.com/MNieddu91/Marlin-AI3M-BLTouch) which has been working great for me so far.

closed 2023-05-28 by faultyDaantje

knutwurst, 2023-05-28:
Can you try to disable the highspeed-Mode?

Just send M401 S0 to the printer and M500 and test again. Maybe that's the issue with your probe.

faultyDaantje, 2023-05-31:
That was it, thank you for helping me.
I sent you €5,- for the great firmware and support :)