knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #465: [BUG] SD card file list recognition error [ Anycubic Mega X ]

tgkang43, 2023-06-03:
If the SD card is removed and reinserted while the printer is on, an error occurs in recognizing the file list.

knutwurst, 2023-06-03:
What firmware Version? What printer configuration?

tgkang43, 2023-06-03:
I'm sorry I didn't provide enough information. The firmware used is MEGA_X_DGUS_v1.5.0-RC1.

knutwurst, 2023-06-03:
I'm on it :)

knutwurst, 2023-06-03:
It should be fixed in https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/commit/6bc410a0e3e895e0c3b57e74093db7325a3d7110.

To get the fix, you need to [compile the firmware](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/Howto:-Compile-Firmware-with-PlatformIO) or wait til the next release.

tgkang43, 2023-06-03:
Thank you !
I look forward to the next release.

knutwurst, 2023-06-04:
Soooo.... since it is fixed, may I close the issue?

tgkang43, 2023-06-07:
Yes, I will close the issue while waiting for the next version.

closed 2023-06-07 by tgkang43