knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #467: [BUG] BLTouch Leveling X/Y do_blocking_move_to X80.75 Y39.00 Z7.45 not working

SjoerdNLD, 2023-06-08:
### Bug Description

On MEGA_TMC_BLT_10m the X / Y movement is not correct using the special menu auto leveling.
After homing it moves normally to the first position, but all points after that make heavy noise and the bltouch moves only some 5mm instead of the expected 45mm.
This started in all releases after 1.5.0.-beta-1.

### Steps to Reproduce

Use any hex file after 1.5.0-beta-1 on MEGA_TMC_BLT_10.3
Store: M851 X-26.00 Y-29.00 Z-2.19
Start auto leveling via special menu

**Expected behavior:** 
Normal movement
do_blocking_move_to  X36.00 Y39.00 Z7.45
do_blocking_move_to  X80.75 Y39.00 Z7.45
and moving 45mm

**Actual behavior:** 
Heavy noise movement
do_blocking_move_to  X36.00 Y39.00 Z7.45
do_blocking_move_to  X80.75 Y39.00 Z7.45
and moving 5mm max

#### Additional Information
Loading firmware 1.5.0-beta-1 makes it work again normally.

* Provide pictures or links to videos that clearly demonstrate the issue.
[heavy noise movement suring autolevel in firmware 1.5.0 beta 2 and later.webm](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/assets/46634319/2f3e764b-1a98-44e7-8655-4ba9e2b1d134)
[normal movement suring autolevel in firmware 1.5.0 beta 1.webm](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/assets/46634319/ead828dd-9a18-4dea-b564-7a09e567b8ca)

Any help is very welcome.

knutwurst, 2023-06-08:
Unfortunately there is no sound in your .webm videos so I can't hear the motors sound.

However, I have two questions:
1. Have you loaded the defaults of the firmware and then saved them?
2. Can you send me a log output of M503?

Best regards

SjoerdNLD, 2023-06-08:
* I added the audio.

1.    Have you loaded the defaults of the firmware and then saved them? Yes
2.   Can you send me a log output of M503? Sure:
`M503
; Linear Units:
G21 ; (mm)
; Temperature Units:
M149 C ; Units in Celsius
; Filament settings (Disabled):
M200 S0 D1.75
; Steps per unit:
M92 X160.00 Y160.00 Z800.00 E196.00
; Max feedrates (units/s):
M203 X500.00 Y500.00 Z6.00 E80.00
; Max Acceleration (units/s2):
M201 X3000.00 Y2000.00 Z60.00 E10000.00
; Acceleration (units/s2) (P<print-accel> R<retract-accel> T<travel-accel>):
M204 P1500.00 R3000.00 T3000.00
; Advanced (B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_jerk> Y<max_jerk> Z<max_jerk> E<max_jerk>):
M205 B20000.00 S0.00 T0.00 X8.00 Y8.00 Z0.40 E10.00
; Home offset:
M206 X0.00 Y0.00 Z0.00
; Auto Bed Leveling:
M420 S0 Z0.00 ; Leveling OFF
; Endstop adjustment:
M666 Z0.00
; Hotend PID:
M301 P14.89 I0.89 D62.52
; Bed PID:
M304 P251.78 I49.57 D319.73
; Z-Probe Offset:
M851 X-26.00 Y-29.00 Z-2.19 ; (mm)
; Linear Advance:
M900 K0.00
; Filament load/unload:
M603 L538.00 U555.00 ; (mm)
; Filament runout sensor:
M412 S1 ; Sensor ON
ok`

knutwurst, 2023-06-09:
As far s I can see, your values are corrupt!

```
; Steps per unit:
M92 X160.00 Y160.00 Z800.00 E196.00
; Max feedrates (units/s):
M203 X500.00 Y500.00 Z6.00 E80.00
; Max Acceleration (units/s2):
M201 X3000.00 Y2000.00 Z60.00 E10000.00
; Acceleration (units/s2) (P R T):
M204 P1500.00 R3000.00 T3000.00
```

These values are way too high! How is it even possible? 
Please issue the commands M502 and M500 afterwards and then look for the values mentioned above.

SjoerdNLD, 2023-06-10:
Too high, thx, I did not see that.
I put 1.5.1, please see the result below.

What sort of value do you recommend, I'll put them.

> M502
//action:notification Hardcoded Default Settings Loaded
Hardcoded Default Settings Loaded
ok
M500
Settings Stored (744 bytes; crc 44690)
//action:notification Settings Stored
//action:notification Settings Stored
ok
M503
; Linear Units:
G21 ; (mm)
; Temperature Units:
M149 C ; Units in Celsius
; Filament settings (Disabled):
M200 S0 D1.75
; Steps per unit:
M92 X80.00 Y80.00 Z400.00 E98.00
; Max feedrates (units/s):
M203 X500.00 Y500.00 Z6.00 E80.00
; Max Acceleration (units/s2):
M201 X3000.00 Y2000.00 Z60.00 E10000.00
; Acceleration (units/s2) (P<print-accel>R<retract-accel> T<travel-accel>):
M204 P1500.00 R3000.00 T3000.00
; Advanced (B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_jerk> Y<max_jerk> Z<max_jerk> E<max_jerk>):
M205 B20000.00 S0.00 T0.00 X8.00 Y8.00 Z0.40 E10.00
; Home offset:
M206 X0.00 Y0.00 Z0.00
; Auto Bed Leveling:
M420 S0 Z0.00 ; Leveling OFF
; Endstop adjustment:
M666 Z0.00
; Hotend PID:
M301 P12.28 I0.75 D50.06
; Bed PID:
M304 P251.78 I49.57 D319.73
; Z-Probe Offset:
M851 X-2.00 Y-25.00 Z-0.40 ; (mm)
; Linear Advance:
M900 K0.00
; Filament load/unload:
M603 L538.00 U555.00 ; (mm)
; Filament runout sensor:
M412 S1 ; Sensor ON
ok

knutwurst, 2023-06-10:
You don't need any recommended values, because every firmware is already delivered with the best settings for the specific printer. :)

SjoerdNLD, 2023-06-11:
Ok I understand, but the values are the same.
After M502 // M500 the lines which you indicated above (M203/M201/M204) are not changed.

Please see serial log of connection, I used the MEGA_TMC_BLT_10_v1.5.1.hex 

> start
Marlin 2.1.2.1
echo: Last Updated: 2023-06-09 | Author: (knutwurst)
echo: Compiled: Jun 10 2023
echo: Free Memory: 1446  PlannerBufferBytes: 1536
//action:notification Knutwurst's Anycubic Printer Ready.
echo:V88 stored settings retrieved (744 bytes; crc 44690)
//action:notification Stored settings retrieved
echo:; Linear Units:
echo:  G21 ; (mm)
echo:; Temperature Units:
echo:  M149 C ; Units in Celsius
echo:; Filament settings (Disabled):
echo:  M200 S0 D1.75
echo:; Steps per unit:
echo:  M92 X80.00 Y80.00 Z400.00 E98.00
echo:; Max feedrates (units/s):
echo:  M203 X500.00 Y500.00 Z6.00 E80.00
echo:; Max Acceleration (units/s2):
echo:  M201 X3000.00 Y2000.00 Z60.00 E10000.00
echo:; Acceleration (units/s2) (P<print-accel> R<retract-accel> T<travel-accel>):
echo:  M204 P1500.00 R3000.00 T3000.00
echo:; Advanced (B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_jerk> Y<max_jerk> Z<max_jerk> E<max_jerk>):
echo:  M205 B20000.00 S0.00 T0.00 X8.00 Y8.00 Z0.40 E10.00
echo:; Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:; Auto Bed Leveling:
echo:  M420 S0 Z0.00 ; Leveling OFF
echo:; Endstop adjustment:
echo:  M666 Z0.00
echo:; Hotend PID:
echo:  M301 P12.28 I0.75 D50.06
echo:; Bed PID:
echo:  M304 P251.78 I49.57 D319.73
echo:; Z-Probe Offset:
echo:  M851 X-2.00 Y-25.00 Z-0.40 ; (mm)
echo:; Linear Advance:
echo:  M900 K0.00
echo:; Filament load/unload:
echo:  M603 L538.00 U555.00 ; (mm)
echo:; Filament runout sensor:
echo:  M412 S1 ; Sensor ON
//action:prompt_end
echo:SD card ok
M502
//action:notification Hardcoded Default Settings Loaded
echo:Hardcoded Default Settings Loaded
ok
M500
echo:Settings Stored (744 bytes; crc 44690)
//action:notification Settings Stored
//action:notification Settings Stored
ok
M503
echo:; Linear Units:
echo:  G21 ; (mm)
echo:; Temperature Units:
echo:  M149 C ; Units in Celsius
echo:; Filament settings (Disabled):
echo:  M200 S0 D1.75
echo:; Steps per unit:
echo:  M92 X80.00 Y80.00 Z400.00 E98.00
echo:; Max feedrates (units/s):
echo:  M203 X500.00 Y500.00 Z6.00 E80.00
echo:; Max Acceleration (units/s2):
echo:  M201 X3000.00 Y2000.00 Z60.00 E10000.00
echo:; Acceleration (units/s2) (P<print-accel> R<retract-accel> T<travel-accel>):
echo:  M204 P1500.00 R3000.00 T3000.00
echo:; Advanced (B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_jerk> Y<max_jerk> Z<max_jerk> E<max_jerk>):
echo:  M205 B20000.00 S0.00 T0.00 X8.00 Y8.00 Z0.40 E10.00
echo:; Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:; Auto Bed Leveling:
echo:  M420 S0 Z0.00 ; Leveling OFF
echo:; Endstop adjustment:
echo:  M666 Z0.00
echo:; Hotend PID:
echo:  M301 P12.28 I0.75 D50.06
echo:; Bed PID:
echo:  M304 P251.78 I49.57 D319.73
echo:; Z-Probe Offset:
echo:  M851 X-2.00 Y-25.00 Z-0.40 ; (mm)
echo:; Linear Advance:
echo:  M900 K0.00
echo:; Filament load/unload:
echo:  M603 L538.00 U555.00 ; (mm)
echo:; Filament runout sensor:
echo:  M412 S1 ; Sensor ON
ok

SjoerdNLD, 2023-06-11:
The default M92 X80.00 Y80.00 Z400.00 E98.00 does not work.
For example, with this setting when I move the X 10mm, it moves only 5mm.
This works normally: M92 X160.00 Y160.00 Z800.00 E196.00
And with extruder feed callibrated:
M92 X160.00 Y160.00 Z800.00 E232.05

SjoerdNLD, 2023-06-11:
If I can do anything, let me know, I like digging for the cause (-:
Funny thing, when I reflash 1.5.0 beta 1, it functions normally.

knutwurst, 2023-06-11:
Why du you need to double the step values?
Are your stepper drivers configured to use 1/32 microsteps? See #313

SjoerdNLD, 2023-06-11:
👏🎉 Very nice, thx!, I did not know that, yes I use TMC1225.
I removed the ms1 pin and put a resistor to gnd.

After modding the tmc1225 I have started auto leveling via special menu, and, working as before, THX!

[image]

closed 2023-06-11 by knutwurst