knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #468: [BUG] Chiron dead after Update to 1.5.0

michaelnum, 2023-06-09:
Hello,

my Chiron Bj 11.2022 is dead after the update to the final version 1.5.0-TMC.
The 1.5.0 Beta 3 was working.

Dead means the display has no connection, it shows only  0 °C, no file menu and no special menu, no beep and the normal Cura does not find it.
Update back works only with the old Cura 15.04.

knutwurst, 2023-06-09:
Hi @michaelnum ,
This can't have anything to do with the firmware since all printers have the same hardware (Trigorilla with Atmgea256). In most cases, the mainboard is "dead" if the firmware was not flashed or not flashed correctly because errors occurred during the transfer. I suspect a coincidence, especially since the beta 3 worked then.

Please do not use Cura, either directly AVRDude (command line tool), or programs that use AVRDude in the background (Visual Studio Code, PrusaSlicer, SuperSlicer, Slic3r etc).

There you will also get an output with a CRC check (which Cura does not do). Here the flash is read out and checked again after the transfer. If this check fails, something is wrong with the hardware. Possibly USB cable, driver etc.

The result should look like this:

[image]

michaelnum, 2023-06-09:
Dear Oliver,

sorry, with the prusaslicer is the same.

[image]

knutwurst, 2023-06-09:
Did you change anything in the printer hardware?

I just checked again what changes I made after Beta 3 and apart from the display and various functional changes I finally adjusted the pins for the filament sensor. Previously this was 19 (like the MEGA) and now it's 33.

However, I have received feedback from several Chiron users that the firmware runs without problems. Both on the old Chriron (with Anycubic 1.0 display) and on the new one (with Anycubic 0.0.2).

If your board doesn't work at all after the flash, it must have something to do with it.

michaelnum, 2023-06-09:
no, this time i made no changes.

In the past i only changed to the TMC-driver and installed an ESP8266-WLAN-modul.

Revert to the 1.5.0-Beta3-TMC (file from Github without WLAN), all is working.

michaelnum, 2023-06-09:
With the  MEGA_S_TMC_v1.5.0.hex there is the same problem:  it shows only 0 °C, no file menu and no special menu, no beep
Additionally with this firmware always the message from filament sensor is shown.

knutwurst, 2023-06-09:
But the MEGA_S_TMC_v1.5.0.hex works on your second printer? 
Of course you get the filament error because the filament sensor pin differs.

There must be something wrong with your hardware.

If I were you, I would open the printer and disconnect all cables from the board except for the display and try again.

Also, als alternative, you can try to use the current master build, which already is updated to marlin 2.1.2.1 for the 1.5.1 release.

michaelnum, 2023-06-09:
Yes, on the normal I3 Mega (without S)  the MEGA_TMC_v1.5.0.hex works.

But if i flash the 1.5.0 beta3, it works. So it can not be a hardware problem.

But i'm confused.
After i flashed the first time the  CHIRON_TMC_v1.5.0.hex, coming from Beta3, i made directly the load defaults and Save EEprom. After the restart there was the problem.

Now i made the same, but now it is differently. 
In Beta 3 i load defaults and Save EEprom.
After the update i can not see the special menu again.

Is the problem with the parameters in the EEprom?

knutwurst, 2023-06-09:
That's pretty strange.
What's the output if you load the defaults and save the eeprom via pronterface? (M502, M500)

michaelnum, 2023-06-09:
That is the status when connecting with 1.5.0-final:

Connecting...
start
Printer is now online.
Marlin 2.1.2
echo: Last Updated: 2023-06-07 | Author: (knutwurst)
echo: Compiled: Jun  7 2023
echo: Free Memory: 1447  PlannerBufferBytes: 1536
//action:notification Knutwurst's Anycubic Printer Ready.
echo:V87 stored settings retrieved (737 bytes; crc 50381)
//action:notification Stored settings retrieved
echo:; Linear Units:
echo:  G21 ; (mm)
echo:; Temperature Units:
echo:  M149 C ; Units in Celsius
echo:; Filament settings (Disabled):
echo:  M200 S0 D1.75
echo:; Steps per unit:
echo:  M92 X80.00 Y100.00 Z400.00 E415.00
echo:; Max feedrates (units/s):
echo:  M203 X100.00 Y100.00 Z20.00 E60.00
echo:; Max Acceleration (units/s2):
echo:  M201 X350.00 Y350.00 Z50.00 E20000.00
echo:; Acceleration (units/s2) (P<print-accel> R<retract-accel> T<travel-accel>):
echo:  M204 P350.00 R2000.00 T350.00
echo:; Advanced (B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_jerk> Y<max_jerk> Z<max_jerk> E<max_jerk>):
echo:  M205 B20000.00 S0.00 T0.00 X4.00 Y4.00 Z0.40 E10.00
echo:; Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:; Auto Bed Leveling:
echo:  M420 S0 Z0.00 ; Leveling OFF
echo:  G29 W I0 J0 Z-1.00000
echo:  G29 W I1 J0 Z-1.00000
echo:  G29 W I2 J0 Z-1.00000
echo:  G29 W I3 J0 Z-1.00000
echo:  G29 W I4 J0 Z-1.00000
echo:  G29 W I0 J1 Z-1.00000
echo:  G29 W I1 J1 Z-1.00000
echo:  G29 W I2 J1 Z-1.00000
echo:  G29 W I3 J1 Z-1.00000
echo:  G29 W I4 J1 Z-1.00000
echo:  G29 W I0 J2 Z-1.00000
echo:  G29 W I1 J2 Z-1.00000
echo:  G29 W I2 J2 Z-1.00000
echo:  G29 W I3 J2 Z-1.00000
echo:  G29 W I4 J2 Z-1.00000
echo:  G29 W I0 J3 Z-1.00000
echo:  G29 W I1 J3 Z-1.00000
echo:  G29 W I2 J3 Z-1.00000
echo:  G29 W I3 J3 Z-1.00000
echo:  G29 W I4 J3 Z-1.00000
echo:  G29 W I0 J4 Z-1.00000
echo:  G29 W I1 J4 Z-1.00000
echo:  G29 W I2 J4 Z-1.00000
echo:  G29 W I3 J4 Z-1.00000
echo:  G29 W I4 J4 Z-1.00000
echo:; Endstop adjustment:
echo:  M666 Z0.00
echo:; Hotend PID:
echo:  M301 P20.00 I0.50 D106.55
echo:; Bed PID:
echo:  M304 P97.10 I1.41 D1675.16
echo:; Z-Probe Offset:
echo:  M851 X0.00 Y0.00 Z-16.80 ; (mm)
echo:; Linear Advance:
echo:  M900 K0.00
echo:; Filament load/unload:
echo:  M603 L538.00 U555.00 ; (mm)
echo:; Filament runout sensor:
echo:  M412 S1 ; Sensor ON
//action:prompt_end
echo:SD card ok
Disconnected.

michaelnum, 2023-06-09:
From 1.3.1 (after flashing Load defaults and save eprom) to 1.5.0 the same problem.

With 1.5.0 i get now answer to my commands. But the connection ist ok, it send the status to the pronterface.
For an M114 i get no answer, too.

I'm waiting for the 1.5.1 release with marlin 2.1.2.1

knutwurst, 2023-06-09:
Is it possible for you to compile the firmware yourself and find out from which commit or at least from which beta it stopped working?

Unfortunately I can't test it myself.

knutwurst, 2023-06-09:
[CHIRON_TMC_v1.5.1-pre.hex.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/11704925/CHIRON_TMC_v1.5.1-pre.hex.zip)

You can try if you want. But this is completely untested code.

michaelnum, 2023-06-09:
... now it works... with removed sd-card
It seems that something on the sd-card the final 1.5.0 firmware does not like. although it said: echo:SD card ok
Oliver, thanks for your fast help

knutwurst, 2023-06-09:
Wait.. what?
And what if you remove and re-insert the sd card?
That's strange indeed. I'll investigate the issue.

knutwurst, 2023-06-09:
I did some refactoring regarding the file list on the sd card. Would you mind testing it?
[CHIRON_TMC_v1.5.1-pre2.hex.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/11705712/CHIRON_TMC_v1.5.1-pre2.hex.zip)

michaelnum, 2023-06-09:
Hello Oliver,
it works only if the sd-card is not inside.
The i3mega have the same problem with this. 
But on the sd-card all looks good. No special letters.
Some gcode-files in the root, some in the gcode-files in the folder "old" and the folder "System Volume Information" with two files inside.

For the i3Mega i compiled the newest github source.
But the problem still exist.

Now i will test to remove some files

michaelnum, 2023-06-09:
I can not simulate it.

But i have seen a problem, but i don't know if that is the same:
If i go up and down in the file menu, and at the 
end one klick to much down or up, it shows nonsense.
On i3Mega with compiled the newest github source.

[image]

michaelnum, 2023-06-09:
ok, i made some test with the beta versions.
The 1.5.0-rc1 is Ok.
I only this problem with the file menu with the final 1.5.0 and newer.

knutwurst, 2023-06-09:
But the RC1 does not work without the SD card, am I right?
Can you test the current master again? I've made some changes regarding string memory management.

michaelnum, 2023-06-09:
Yes, the RC1 have a problem without sd card.

The current master looks good, i see no problem.

Thanks!

knutwurst, 2023-06-09:
Really?!
It works?
So... it was only one simple stringLib I used in the past?

Shame on me. I thought, I might use "strdup()".

Great to hear that it works now and thanks for testing!

closed 2023-06-10 by knutwurst