knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #469: [BUG] (short description)

helloiamdavid, 2023-06-09:
### Bug Description
1-I Can't print from the SD memory.

### Steps to Reproduce

1. I installed firmware : MEGA_S_v1.5.0.hex    I have an Anycubic Mega S, ( the version with 2 Z-endstops) 
2. I completed the setup:
            -►special menu 
            -►Load FW
            -► save
            -►Exit
3. I tried to print an old file, but the didn't appered when I introduced the SD
4. I used a new SD, loaded a file and I could see the file, but when I click on print nothing happens
5. I traied through pronterface and could send to print the file, but I need to keep my laptop conected 

**Expected behavior:** Print from the SD 

**Actual behavior:** I can print only with pronterface [](url)

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

knutwurst, 2023-06-09:
Did you put the SD card directly into the printer, or did you use an SD extension?

Does the SD card work with a different firmware? How is the SD card formatted? What files are on it? What kind of printer are you using exactly? What firmware (filename) are you using? Did you restart the printer after flashing?

michaelnum, 2023-06-09:
Perhaps the same like: https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/issues/468

Do you see the special menu? Are the tempertures OK or all 0 °C?

knutwurst, 2023-06-09:
@michaelnum @helloiamdavid said that he did the initial procedure, so the display works. It's something else.

helloiamdavid, 2023-06-09:
> Did you put the SD card directly into the printer, or did you use an SD extension?
I put the original SD  ( the one included with the printer) directly into the printer with all my gcodes, the printer read the SD or at least it showed the message it did it, but when I click on print I can see only the special menu. 
 also when I connect with pronterface I can send to print from the SD in pronterface, so it means the printer is reading the SD but not showing the files. 
> Does the SD card work with a different firmware? 
Yes, is the one I used with the original firmware
> How is the SD card formatted?
FAT 32
> What files are on it?
Gcodes and the 2 folders included when I bought the printer with PDF's, drivers and Cura software
 > What kind of printer are you using exactly? 
Anycubic Mega S  the version with 2 Z-endstops
> What firmware (filename) are you using?
latest version: MEGA_S_v1.5.0.hex 
I had the original version from factory (Marlin_TFT_MEGA_S_V1.1.5.ino.hex)
> Did you restart the printer after flashing?
Yes, many times.
> Do you see the special menu? Are the tempertures OK or all 0 °C?
Yes I see it, and temp are ok

**I made some tests with a new SD and it worked after restart again the printer, maybe the original memory got damaged in some way that the printer can't display the files,  or maybe there are too many files. its weird, because I can read the SD in my laptop**

I am printing with the new SD 
I'll keep testing the old SD and let you know what I found

knutwurst, 2023-06-09:
There was actually an error in the memory management of the file list. So it may be that this affected you too.
The error is fixed, but it would be great if you could verify this as well.

Can you compile the firmware yourself? There are instructions on the wiki.

That would be really great!

knutwurst, 2023-06-11:
Fixed in 1.5.1.

closed 2023-06-11 by knutwurst