knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #472: [BUG] (short description)

teddibaer, 2023-06-18:
### Z-Offset Issue with Chiron

I installed the latest (1.5.1) version on my Chiron (first time user).
I loaded/stored the firmware defaults and changed Cura G-code init as per wiki.

First print with Cura ended up with the Extruders Z0 being several millis above the plate.
Did auto leveling with the probe -> same result.

After some manual fiddling with the Z-offset I now somewhat got it working (with M206 Z-2.4 - need some more testing though), so it seems there is an offset between auto level data and real world.
(As a side note: the "easy 4 point leveling" seems broken - I can see the 4 points in the menu, but cannot move Z up/down?.)

I am not a marlin expert (or even 3D printer expert), but it seems to me, that the Z-Probe-Offset is wrong - maybe that causes the issue: its set to -18.9 (as per M503), but when measuring it seems to be more like 17.something (I need to measure it more exactly again).

### Steps to Reproduce

1. load firmware defaults (on Chiron)
2. do auto level with probe
3. use Cura init G-code and print something (with Cura)

knutwurst, 2023-06-19:
Hi!
You're the first to make this observation, so I'm not sure if there might be an understanding problem. Are you sure your offset is 17.something? Was it like that in the original firmware too? Basically, it doesn't matter at which value you get a perfect first layer, as long as this value is constant.

The Z offset describes the distance from the sample to the nozzle. Namely from the point at which the probe triggers (not the length of the probe or similar)

Also, you don't have to change anything in the terminal if your offset isn't correct. I have implemented the feature in such a way that you can also adjust the offset AFTER an autoleveling. At the same time, all points of the mesh are shifted by this value, which enables perfect printing.

Incidentally, the 4-point easy leveling is nothing more than approaching the corners of the bed. It is used to manually align the bed correctly. It's not a feature to change any offsets. It is only an aid and is available with every printer - even if you don't use mesh leveling at all.

I hope this helps you a bit and I look forward to updates :)

teddibaer, 2023-06-19:
Thx for the quick response. It might very well be, that I am doing something (very) wrong. 

I checked the Anycubic Chiron 1.3 sources, and there it is defined as 
#define Z_PROBE_OFFSET_FROM_EXTRUDER -14

I will continue fiddling later today probably ...

knutwurst, 2023-06-19:
The original Anycubic Z-Offset is extremely far from reality. You can't rely on that.

You could of course do the comparison, install the stock firmware, level up until it fits perfectly (with the display!) and then read the Z offset.

Still, I wouldn't count on it.

knutwurst, 2023-06-19:
> I have my doubts that the hardware has zero tolerance and that there is something like a one-fits-all value. Well, -2.4 is quite a difference, but if the actual problem can be resolved settings the offset once that should not be a big deal anyway.

That's what I thought ;) 

@teddibaer forget about M204 and set your Z Offset to the value that works for you. If -18.9 works, just use it. It seems not that far away from the current default of -16,8 (which is the value I tested in the past. And as you can see... it's also far away from anycubic's -14.

All printers and probes are different. That's the reason you can set it in the firmware.

knutwurst, 2023-06-19:
-18.9 is not the default, but your currently configured value. If you load the FW defaults, it resets to -16,8.

When you change the value in the Leveling menu, it stores your settings inside the EEPROM which is loaded on every single boot.

teddibaer, 2023-06-19:
Ok. I am not sure what happened and why my Z-Offset was set to -18.9.  After again loading the firmware defaults it indeed was set to -16.8.
However, this also did not really work for me - after several tests I believe that -16.0 is about right for my machine.

I still seem to have some issues with underextrusion, but that is a different topic.

Thx for being super helpful in clearing things up for me (noob).

closed 2023-06-19 by knutwurst