knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #478: model predictive control

Ingegneus, 2023-06-25:
today I wanted to set up the model predictive control of marlin, but sending M306 T resulted in "Unknown command "M306 T"". since the FW should be at 2.1.2.1 of marlin, shouldn't this feature be present already? and if not, would it be possible to implement?

knutwurst, 2023-06-25:
MPCTEMP is an experimental feature which is not enabled by default. You need to compile the firmware by yourself and test ist.

See: https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/blob/160575dd4f158cae7cdc8aa7289fa42efb6623eb/Marlin/Configuration.h#L751

And: https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/Howto:-Compile-Firmware-with-PlatformIO

closed 2023-06-25 by knutwurst

Ingegneus, 2023-06-25:
I should have mentioned this in the original post, that I enabled it and disabled PID control and that gave me the result I described