knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #479: [FR] Support for Gcode M81 - Power Off

ChristophDerndorfer, 2023-06-25:
### Description

I'm using FW version 1.5.1 on a Mega S connected to OctoPrint 1.9.0 and stumbled across the [PSU Control plug-in](https://plugins.octoprint.org/plugins/psucontrol/) that allows an automatic shutdown of a 3D printer with various methods. One of them is using Marlin's command [M81 - Power Off](https://marlinfw.org/docs/gcode/M081.html). I tried manually sending the command and get an "Recv: ok" but since nothing happens, I assume that the command either isn't supported by the firmware or the printer itself.

### Feature Workflow

1. Support for the M81 command so that the PSU can be shut off via Gcode.

#### Additional Information

* Nothing to add here.

stklcode, 2023-06-26:
The Mega S PSU has a mechanical switch and a fixed power output. No relay, power-enable-pin (like typical computer PSUs have) or similar connected to the board.

So unless you modify the hardware, you it is impossible to use M80/M81 with this machine.

ChristophDerndorfer, 2023-06-26:
Got it, thanks for the quick response and insights.

closed 2023-06-26 by ChristophDerndorfer