knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #481: (minor error) slightly wrong file path in wiki for location of compiled firmware

kgabriel-dev, 2023-07-15:
### Description

To help solving issue #476 I wanted to compile the firmware myself with the code in the current `master` branch. After reading your documentation on how to do this, it was very easy for me to compile the firmware myself. The only problem I noticed is that the `.hex` file was not at the described location. The location in the wiki is `/FIRMWAREFOLDER/Marlin/.pio/build/[VARIANT]/firmware.hex`. However, at least for me, the file is here: `/FIRMWAREFOLDER/.pio/build/[VARIANT]/firmware.hex`. In short, the folder `.pio` is in the root directory of the project and not in the subfolder `Marlin`. As this is a very minor mistake in the wiki, I didn't want to create an extra pull request for that fixing the issue. The image in the wiki shows the correct path missing the `/Marlin/` in it.

GammaGallagher, 2023-07-15:
This caught me out.  Thanks for sharing.

knutwurst, 2023-07-15:
Oh, Thanks!
I've edited the paths ;)

closed 2023-07-15 by knutwurst