knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #482: MEGA_P_DGUS_BLT_10_v1.5.2 with input shaping

djepsylon, 2023-07-23:
### Description

I ve compiled a build of the MEGA_P_DGUS_BLT_10_v1.5.2 with the input shaping function operational.
i give you the hex file for somes poeple who wants and can't compile it himself.
i've just added this feature all is like @knutwurst have set the firmware.
i ve tested it and it works well in my i3mega pro
[firmware.zip](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/12137517/firmware.zip)

knutwurst, 2023-08-01:
You can simply reduce the usb/serial input buffer sizes: https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/blob/97c3391cd9c3f9e12983ebe618891c65b2f95e9e/Marlin/Configuration_adv.h#L2417

Also you can disable the sd card file sorting to save more ram: https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/blob/97c3391cd9c3f9e12983ebe618891c65b2f95e9e/Marlin/Configuration_adv.h#L1614

I haven't had time to work with InputShaping before, but would appreciate your results

djepsylon, 2023-08-04:
thanks a lot i will try this changes and i tell you if i work or not i ve not a lot of time too

schmttc, 2023-08-16:
Be aware, I compiled and flashed 1.5.2 with added input shaping on a Mega S Trigorilla 0.0.2 (code: https://github.com/schmttc/Marlin-2-0-x-Anycubic-i3-MEGA-S/tree/local) . The compiler advised 99.4% RAM use. The LCD came up, but I could not connect over serial USB, and it kept beeping as if the board was resetting in the background.
Also the SD card did not list contents in the menu, but that may be a separate issue #486 

I had to recover my board by re-flashing the bootloader and uploading an older firmware, I used this guide https://cyburdine.wordpress.com/2022/07/18/trigorilla-bootloader-and-firmware-install-process/

The author of Input Shaping advises to have ~1Kb RAM free https://reprap.org/forum/read.php?415,891975,892852#msg-892852
And so we will need to diable some other features to use Input Shaping on boards with ATMEGA2560

djepsylon, 2023-08-16:
@schmttc  hi, yes this is what i ve see too. impossible to put input shaping in he trigorilla bord due to under memory space. 
i think if we want to use this option with i3mega series, we must change the motherboard, no other choices.
after comment all other fuctions for can compile input shaping results in there is only input shaping active lollll ==>>no benefits.

schmttc, 2023-08-16:
Hey @djepsylon , try leaving all features enabled, enable Input Shaping, and uncomment "#define SHAPING_MAX_STEPRATE 10000" in Configuration_adv.h. 
My compiler then says 7018 of 8192 bytes RAM, which may be enough overhead to allow things to run correctly. I won't have time to test this until the weekend.

djepsylon, 2023-08-17:
hi @schmttc , thanks to you
 i ve truied your solution it works fine but it seems there is some problems for set input shaping correctly, i don't know why for the moment i don't have tme to test more deepest for now, maybe next week if you don't have test yourself in this time or another poeple

schmttc, 2023-08-21:
The firmware compiles and runs fine, @knutwurst let me know if you'd like me to raise a PR for the functionality.

I spent more time than I wanted to testing and trying to dial it in, but I am starting to agree with @djepsylon that the MEGA2560 is too limited to take advantage of this feature. I couldn't get a substantial speed increase without a substantial decrease in quality.

djepsylon, 2023-08-21:
Effectively, due to lot of memory used, the motherboard seems to pain to do work correctly and don't give good results. the benefits are to small or not present. i think that, with the mega2560, the print results are very good without input shaping and activate this option make the performances lower and print quality too. like @schmttc says, this mainboard seems to be too limited for use input shaping but the printer already make very good prints without it

closed 2024-06-05 by knutwurst