knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #483: Mega S not using Manual Mesh Leveling

claversky, 2023-08-01:
### Bug Description

Hi everyone! Kinda new to all this but since my Mega S has a warped bed in the middle and I'm unable to get a proper first layer, I did found that by using this firmware and mesh bed leveling option would allow me to correct this. So I did flash the firmware (1.5.2) with the correct one in the list, Load EEPROM and Saved Settings. Then I used the Easy Leveling to get the 4 corners aligned as best as I could and then started the Mesh Leveling, saving EEPROM as well. When sending the proper commands to PronterFace after restarting, I can confirm that the Mesh was saved. Also, added the Start G-Code provided here. After many attempts to get it to work properly, I'm starting now to assume that when my printer home by using the Z Endswitches, it doesn't take in mind the mesh bed leveling (even tough apparently everything is setup correctly) since the printer still uses apparently the Easy Leveling and doesn't go beyond the Endstop to correct for the warped bed, which causes not being able to get a first layer. I'm open to any suggestions and troubleshooting offered, since I searched everywhere and I'm unable to understand really how to solve this. BTW, printing with SD Card with only one file on it at the time.

### Steps to Reproduce

Happens in every single print, seems like the Manual Mesh Leveling is not being used at all despite having (hopefully) everything setup correctly.

Expected behavior:** Mesh Bed Leveling being used to get a proper first layer adhesion

**Actual behavior:** Mesh Bed Leveling is not being used, which causes to PLA to not adhere properly to the Bed

Start G-Code in Cura:
G21                                        ; metric values
 G90                                        ; absolute positioning
 M82                                        ; set extruder to absolute mode
 M107                                       ; start with the fan off
 M190 S{material_bed_temperature_layer_0}   ; preheat and wait for bed
 M109 S{material_print_temperature_layer_0} ; preheat and wait for hotend
 M300 S1000 P500                            ; BEEP heating done
 G28 X0 Y10 Z0                              ; move X/Y to min endstop
 M420 S1
 G0 Z0.15                                   ; lift nozzle a bit
 G92 E0                                     ; zero the extruded length
 G1 X50 E25 F500                            ; Extrude 25mm of filament in a 5cm line.
 G92 E0                                     ; zero the extruded length again
 G1 E-2 F500                                ; Retract a little
 G1 Y120 F4000                              ; Quickly wipe away from the filament line

claversky, 2023-08-01:
I have a stock Mega S with the DGUS Display so I installed this firmware: [MEGA_S_DGUS_v1.5.2.hex](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/releases/download/1.5.2/MEGA_S_DGUS_v1.5.2.hex)

Hereux, 2023-08-16:
I've had the same thing for months now, but would like to print again. Did you get it repaired already?

claversky, 2023-08-16:
I put a PEI on top of a glass bed and level the Bed literally until the paper barely moves so it squishes the PLA and it’s kinda resolved but in terms of having the manual mesh leveling not yet. Still have some issues on some parts of the bed but I’m limiting prints to the parts that I know Will work properly.

TheRealSaeba, 2023-08-19:
Inreresting. This seems to be the same problem I am currently confronted with.

nCoderGit, 2023-09-25:
You might need to specify the index of the mesh you want to load.

Try `M420 L0 S1` (instead of just `M420 S1`) in your Start G-Code.

If it still doesn't work, my guess is that there might be another G28 call somewhere and your fw is compiled without `ENABLE_LEVELING_AFTER_G28` or `RESTORE_LEVELING_AFTER_G28`. 
(G28 (Auto Home) disables bed leveling by default)

mlewis-everley, 2023-10-18:
@claversky Out of curiosity, how do you know the mesh is not being used? I usually check by putting a finger on one of the Z lead screws while the printer is putting down a first layer. If the screw is turning (or more likely just vibrating ever so slightly) then that should mean the mesh is being applied and your issue might be something else.

Also, I don't know if it makes a difference, but instead of using the following:

```
G28 X0 Y10 Z0 ; move X/Y to min endstop
M420 S1
```

I use this:

```
; Home head and load mesh
G28
M420 S1

; Park nozzle 
G1 Z0.2
G1 Y10
```

TheRealSaeba, 2023-10-18:
> @claversky Out of curiosity, how do you know the mesh is not being used? [...]

I do it similarly. I have attached a 3 cm strip of adhesive film to the lead screw in order to be able to see tiny movements better 

This is part of my start-code:
G28 X0 Y10 Z0  
M420 S1 Z2.0  

Somehow, it works now. I think it was important to perform "Save EEPROM" in the special menu after performing the manual bed levelling. Otherweise the values would only exist until the next reset.

claversky, 2023-10-18:
Hey Everyone and thank you for all the replies!

I will try that @nCoderGit thank you for the help.

In terms of making sure that the mesh bed leveling was not working, I basically created a mesh but with opposite values (in some instances making the nozzle way higher from the bed and sometimes lower) and I realized that the nozzle will consistently just maintain the height, without detecting any changes.

Thanks again and I will keep you guys updated whenever I try things out!

mlewis-everley, 2023-10-18:
> Somehow, it works now. I think it was important to perform "Save EEPROM" in the special menu after performing the manual bed levelling. Otherweise the values would only exist until the next reset.

Yes, you need to save to EEPROM after performing mesh leveling. To my understanding, the mesh is wiped from memory when you home (`G28`) unless you have this disabled in firmware. If this is the case, I think the mesh data needs to be restored from the EEPROM when a `M420` is called (thought I could be wrong).

closed 2024-06-05 by knutwurst