knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #486: 4MAX Pro SD card not shown on the display

tk61, 2023-08-14:
### Bug Description

Since 1.5.x the SD card content is not shown on the screen. - V1.4.4 is working

### Steps to Reproduce

1. Flash any V1.5.x
2. LOAD DEFAULT / SAVE
3. Insert SD card 
4. Goto "Print" - Special Menue will be shown but not the SD card content - GCODE M20 is working
5. POWER OFF / ON then goto 3., its the same 

#### Additional Information

* SD card currently formated with FAT
* copied some *.gcode files on it, also 8.x filenames

tk61, 2023-08-17:
Ich habe das Verhalten bezüglich der SD_Karte selbst etwas genauer untersucht und festgestellt:
1. In pins_TRIGORILLA_14.h den SD_DETECT_PIN ändern auf nicht vorhanden
    _#if ANY(ANYCUBIC_LCD_CHIRON, ANYCUBIC_LCD_I3MEGA, ANYCUBIC_TOUCHSCREEN) // PATCH: knutwurst
        #ifndef BEEPER_PIN
          #define BEEPER_PIN               EXP2_03_PIN  // Chiron Standard Adapter
        #endif
        #define SD_DETECT_PIN              -1  // EXP2_07_PIN  // Chiron Standard Adapter //PATCH PATCH by tk - EXP2_07_PIN -> -1
      #endif_

Jetzt erscheint der SD-Karten-Inhalt im Display, aber nach dem Entfernen oder Tauschen der SD-Karte bleibt der alte Inhalt erhalten. Deshalb habe ich

2. In Marlin\src\lcd\extui\knutwurst\anycubic_touchscreen.cpp bei der Funktion RenderCurrentFileList zwei Zeilen hinzugefügt

void AnycubicTouchscreenClass::RenderCurrentFileList() {
  
card.release();                                         //SD workarount, added by tk
  card.mount();                                           //SD workarount, added by tk_

  currentFileOrDirectory[0] = 0;             
    uint16_t selectedNumber   = 0;
  FileList currentFileList;

Jetzt wird auch der Inhalt beim Wechsel der SD aktualisiert (auf Knopfdruck).
Das Ganze ist jetzt bestimmt ziemlich hemdsärmlich, aber ich bin kein Programmierer und es funktioniert erst mal so für mich.

knutwurst, 2023-08-17:
Hi @tk61 ,
Oha! Das hatte ich noch gar nicht auf dem Schirm. Gute Arbeit!

Darf ich fragen, welchen Drucker du genau hast? Ich gehe mal davon aus, dass du einen 4MAX Pro als "nicht 2.0" Version hast, korrekt? Den konnte nämlich bisher niemand testen. Der 4MAX Pro 2.0 funktioniert, aber offensichtlich nutzt deiner einen anderen SD_DETECT_PIN, oder deiner ist defekt. 

Das müsste m.M.n. auch der einzige Fehler sein. Das card.release()/card.mount() sollte eigentlich überflüssig sein. Denn wenn eine SD Karte erkannt ist, müsse das Mounting dennoch stattfinden. Dazu habe ich hier eine Änderung für SD Extender eingebaut, die den Pin blockieren und immer "eingelegt" zurückgeben:
https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/blob/97c3391cd9c3f9e12983ebe618891c65b2f95e9e/Marlin/src/lcd/extui/knutwurst/anycubic_touchscreen.cpp#L1439

Das ist in der Refresh-Routine und mounted die SD-Karte, wenn kein Inhalt zu sehen ist, aber der SD_DETECT_PIN belegt ist.

Es könnte also zu Problemen in deiner Lösung kommen, wenn keine SD Karte eingelegt ist. Mindestens auf der Konsole müsste es Fehler hageln, weil das mount() fehlschlägt, welches du ja erzwingst.

Du bist auch der Erste, der von diesem Fehler berichtet, weshalb ich mir vorstellen könnte, dass dein Kartenleser eventuell defekt ist und der SD_DETECT_PIN nicht auslöst. Durch das Setzen auf -1 wird der Pin gar nicht mehr ausgelesen.

Dass dies in der 1.4.4 noch funktionierte liegt daran, dass ich hier das mounting auch erzwungen habe - auch wenn es extrem unschön war. ;)

tk61, 2023-08-17:
Danke Oliver für deine ausführlichen Erläuterungen.

Dass vielleicht mein DETECT Pin kaputt ist, auf die Idee bin ich überhaupt nicht gekommen.

Würde aber am ehesten Sinn machen. Ich werde die Tage wieder am Drucker basteln, dann schau ich nach dem Detect-Pin.

Ich melde mich wieder, wenn ich mehr weiß.

Und ja, ich habe einen reinen 4MAX Pro.

 

Grüße

Thomas

 

https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/blob/97c3391cd9c3f9e12983ebe618891c65b2f95e9e/Marlin/src/lcd/extui/knutwurst/anycubic_touchscreen.cpp#L1439

tk61, 2023-08-18:
Hallo Oliver,
ich habe heute nach dem SD_DETECT Pin geschaut und siehe da, der konnte nie funktionieren, weil der SD-CARD Slot schon in der Fertigung falsch zusammengebaut war. An dem Abdeckblech ist ein Kontakt und der hatte immer Kontakt zu dem SD_DETECT Pin. 
Blech richtig montiert - Orginal V1.5.2 geflasht und Alles funktioniert, wie es soll.

Vielen Dank für deine kompetente Hilfe und bei dieser Gelegenheit auch gleich mal ein "Vielen, vielen Dank!" für deine/eure Bemühungen rund um die Anycubic Firmware.

Thomas

closed 2023-08-18 by tk61