knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #487: [FR] (Sorry if I am posting this in wrong category)

StrikerP75, 2023-08-25:
Hi I am totally new to github and I have question about what HEX file I should use for Trigorilla Board Model 4 this one
[image]
I am really sorry if I am causing some problems but I am just total newbie to 3D printing and to be honest I have no freaking idea about building marlin.

Ingegneus, 2023-09-28:
read the beginners guide here: https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/Beginner's-Guide-(English)

if you have any specific questions after that, feel free to ask

knutwurst, 2023-10-12:
Hi @StrikerP75 ,
yout board is something completely different. It uses the CHITU firmware and does not support Marlin afaik.

closed 2023-10-12 by knutwurst