knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #490: [BUG] (short description)

EnableDevice, 2023-09-20:
### Bug Description

Hello.

When using the M600 command, the "Continue" button appears on the screen only on the first call. If there are two plastic changes in the code, then for the second time the “Pause” and “Stop” buttons are present on the screen. And further printing is impossible.

#### Additional Information

As I understand it, this is a known problem, since another author found this in releases: "Fixed M600 issue where the screen only showed continue on the first time it was called, not on any further calls".

knutwurst, 2023-10-12:
Good finding. This might be a bug. I'll take a look. Thanks!

EnableDevice, 2023-10-12:
Hello, Oliver.

This issue occurred in build "MEGA_S_v1.5.2". I'll attach a photo of what the display should have looked like during a normal pause, and what it actually looked like. Now I have installed TMC drivers, 3DTouch and rebuilt the firmware. I'll see how the printer behaves now.

[image]
[image]

EnableDevice, 2023-10-12:
I conducted a series of experiments and found out an interesting thing! The M600 command is not just executed once in the code, it is executed once at all! That is, if after flashing the firmware you run the code with this command, it will be executed! The plastic is unloaded, the printer beeps, and the Pause button changes to a Continue button. And you can continue printing. But if you run this code again, the Pause button will not change to a Continue button. At the same time, the plastic is also unloaded, the hotend is parked and the printer beeps. And when you press pause, this is the reaction. Now you can only stop printing. And most importantly, turning off the printer does not produce any results; The M600 command is used only once. :)

[image]
[image]

EnableDevice, 2023-10-12:
So the M600 command got back to work. :) Some kind of floating problem, I can’t understand the pattern yet. I'm monitoring the terminal to see what's being transmitted...

EnableDevice, 2023-10-12:
While I'm poking around in the code, trying to understand why the printer behaves this way, I need to print. And I found a way where all plastic changes are processed correctly. This is printing via OctoPrint. Yes, it's a crutch, but it works. :)

[image]
[image]

maxpd1, 2023-11-15:
Same finding: tried a print with 2 filament changes.
The one and the same GCode behave **always** different.

Either it does not start at all after it is up to temperature or

the pause button does not change to continue when changing the filament or

after a continue, it does not start again, heats only 10 degrees too high, then 10 degrees too low, always alternating or

the temperature display changes to 0 and it cools down completely.

A resume is also not possible, as the target temperature is always 0.

apmyp1990, 2024-01-11:
> While I'm poking around in the code, trying to understand why the printer behaves this way, I need to print. And I found a way where all plastic changes are processed correctly. This is printing via OctoPrint. Yes, it's a crutch, but it works. :)
> 
> ![2023-10-13_00-12-03](https://private-user-images.githubusercontent.com/49840772/274679710-53a18677-2db4-408c-9865-bd7ce95f2a2f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ5ODAwMjQsIm5iZiI6MTcwNDk3OTcyNCwicGF0aCI6Ii80OTg0MDc3Mi8yNzQ2Nzk3MTAtNTNhMTg2NzctMmRiNC00MDhjLTk4NjUtYmQ3Y2U5NWYyYTJmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTExVDEzMjg0NFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTU3NGJiZmJjNmFlZDJjMGUyY2Q1OTFjZjU1NmFkOGQxMjdkY2U4NjRkZDk4YWVkZDI3NTkxNThmZWVlNmMwMDEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.0s8POQeQUfw3Ku1LZbyyuE9R9lmUA7a0bWFBm6hyKzw) ![2023-10-13_00-23-54](https://private-user-images.githubusercontent.com/49840772/274679735-4ab12592-1a4e-4860-afa1-707ce6d68d77.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ5ODAwMjQsIm5iZiI6MTcwNDk3OTcyNCwicGF0aCI6Ii80OTg0MDc3Mi8yNzQ2Nzk3MzUtNGFiMTI1OTItMWE0ZS00ODYwLWFmYTEtNzA3Y2U2ZDY4ZDc3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTExVDEzMjg0NFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTBmMDBlNjViOTg3MzI3MjE1MDIzMzlkY2JiNDU2NTdhYWU0ZTA1YjhjNTM1ZTI5ODBkZGUxMTFjOWFhNjE2ZTkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.totPx5_XILsu7gYSxu4E-ai3w4j6MrZ9h7KgJNa9NbI)

How did you get the color change working in octoprint?

EnableDevice, 2024-01-11:
**apmyp1990**, just a file launched for printing via OctoPrint normally executes the M600 command. The printing stops, the plastic is unloaded and the “Continue” button appears. After pressing it, printing continues as expected.

[image]

apmyp1990, 2024-01-11:
@EnableDevice 
I have printed via octoprint, but there was no message from the printer. Maybe not in the app, I will try again and have a look vie browser. Maybe I missed it there - looging from a smortphone xD

EnableDevice, 2024-01-11:
@apmyp1990
First of all, make sure that the code contains the M600 command on the desired layer. Some slicers install it themselves, you just need to specify the layer, but I write it myself.

```g-code
; layer 25, Z = 3.650
M600
; feature inner perimeter
G1 Z3.650 F1002
...
```

uwetaz, 2024-05-13:
Hello,

I got exactly the same problem as user EnableDevice describes in detail (sporadic no continue button at M600 at nearly 50% of the print jobs).
I have the same old school printer (Mega S with Anycubic 1.0 display) and use version 1.5.3.
The problem also occurs sporadically with always exact same gcode file (I print some Transmission Distance Test Squares for HueForge where M600 is used).
It makes no difference if the printer was switched off before the print or not.

Today I tried a successfull workaround after occuring the problem. After reading here that octoprints „Continue“ helped I read about Marlin M108. So I printed with active console (Pronterface; other terminal SW should also work; I tried TeraTerm) and there also M108 was mentioned („echo:Insert filament and send M108“).
Sending M108 after filament change helps as work arround. So the FW does not freeze (cyclic echo does also not assume a freeze).
But I have to connect via terminal program before the print. Connecting via terminal during print stops the job.

Perhaps it helps to isolate the problem a little bit: Also with sporadic occurence: The display content at the end of print job is also not correct. The end G-code (from wiki here) let the printer beep and normally on display a separate grey summarize window with „ok“ Button cames up with „Printing done took xyz…“. Also in 50% of jobs the mask does no occure after the beeps. If I pressing the STOP button on display and confirm with yes the summarize window occurs. But there seems no dependency if the mentioned M600 problem occured before or not.

Also I observed in case of the problem with M600 the nozzle temperature cames from a higer value (210 instead 200 as usual).

Another problem (It seems not have any influence): Preheat stops after disconnecting pronterface. If I select preheat via menu after disconnecting the set values came only some seconds and after that they reset to Zero. If I connect pronterface again and select preheat via printer menu it works. If I restart the printer it always works.

How can I help to isolate the problem? As revenge to Olivers great work I could spend some time to the issue. For faster way of me we also could exchange private messages in german.
Shall I insert some debug messages into the code or measure at display interface? Is there any good description somewere about the display? I read somewhere about that it sends G-codes to the mainboard. In the meantime I will dig into the code or print a simple cube and test if triggered filament sensor (for easy and fast M600) can provoke the same issue.
I could also test some code changes if someone has any bug fix idea (e.g. via git diff file).

Regards
Uwe

uwetaz, 2024-05-13:
Here I post terminal output in bad and good case of not occured Contiunue button. But it seems not containing enough details.
1) not occured continue button at M600
`echo:Now fresh file: tdsqua~1.gco
File opened: tdsqua~1.gco Size: 175248
File selected
//action:notification TD Square v1.3.gcode
//action:resume
//action:prompt_end
//action:prompt_begin Resuming SD
//action:prompt_button Dismiss
//action:prompt_show
//action:notification TD Square v1.3.gcode
//action:notification Bed Heating...
//action:notification TD Square v1.3.gcode
//action:notification E1 Heating...
//action:notification TD Square v1.3.gcode
echo:Bed Leveling ON
echo:Fade Height OFF
//action:paused
echo:Insert filament and send M108
//action:prompt_end
//action:prompt_begin Nozzle Parked
//action:prompt_button Continue
//action:prompt_show
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
>>> m108
SENDING:M108
//action:prompt_end
//action:prompt_begin Purging...
//action:prompt_button Continue
//action:prompt_show
//action:prompt_end
//action:notification Print Paused
//action:resumed
//action:notification TD Square v1.3.gcode
//action:notification Bed Cooling...
//action:notification TD Square v1.3.gcode
//action:notification Bed Cooling...
//action:notification TD Square v1.3.gcode
//action:notification TD Square v1.3.gcode
//action:notification 20m 18s
echo:Print time: 20m 18s
Done printing file
//action:cancel
//action:notification Print Aborted
echo: cold extrusion prevented`

2) occured continue button at M600
`echo:Now fresh file: tdsqua~1.gco
File opened: tdsqua~1.gco Size: 175248
File selected
//action:notification TD Square v1.3.gcode
//action:resume
//action:prompt_end
//action:prompt_begin Resuming SD
//action:prompt_button Dismiss
//action:prompt_show
//action:notification TD Square v1.3.gcode
//action:notification Bed Heating...
//action:notification TD Square v1.3.gcode
//action:notification TD Square v1.3.gcode
echo:Bed Leveling ON
echo:Fade Height OFF
//action:paused
echo:Insert filament and send M108
//action:prompt_end
//action:prompt_begin Nozzle Parked
//action:prompt_button Continue
//action:prompt_show
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
//action:notification Print Paused
//action:resume
//action:prompt_end
//action:prompt_begin Purging...
//action:prompt_button Continue
//action:prompt_show
//action:prompt_end
//action:notification Print Paused
//action:resumed
//action:notification TD Square v1.3.gcode
//action:resume
//action:notification TD Square v1.3.gcode
//action:notification Bed Cooling...
//action:notification TD Square v1.3.gcode
//action:notification Bed Cooling...
//action:notification TD Square v1.3.gcode
//action:notification TD Square v1.3.gcode
//action:notification 19m 27s
echo:Print time: 19m 27s
Done printing file
//action:cancel
//action:notification Print Aborted
echo: cold extrusion prevented`

uwetaz, 2024-05-17:
Test with triggered filament sensor (for easy and fast M600) can provoke the same issue (sometime button occures sometime not).
Trying to activate LCD_DEBUG in code was not successfull. I built MEGA_S_TMC_BLT_11 on v1.5.3
Result: linker errors
change:
`--- a/Marlin/Configuration.h
+++ b/Marlin/Configuration.h
@@ -3620,10 +3620,10 @@
 //
 // Touch-screen LCD for Anycubic printers
 //
-// #define ANYCUBIC_LCD_I3MEGA
+#define ANYCUBIC_LCD_I3MEGA
 // #define ANYCUBIC_LCD_CHIRON
 #if EITHER(ANYCUBIC_LCD_I3MEGA, ANYCUBIC_LCD_CHIRON)
-  // #define ANYCUBIC_LCD_DEBUG
+        #define ANYCUBIC_LCD_DEBUG
   // #define ANYCUBIC_LCD_GCODE_EXT  // Add ".gcode" to menu entries for DGUS clone compatibility
 #endif`

uwetaz, 2024-05-18:
I found the right define for more verbosity on serial port: ANYCUBIC_TFT_DEBUG
But unfortunately in both cases (Continue button occures or not) the output is the same:
_TFT Serial Debug: FilamentRunout triggered...
TFT Serial Debug: Blocking filament prompt... J23
\//action:prompt_end
\//action:prompt_begin FilamentRunout T0
\//action:prompt_show
\//action:paused filament_runout 0
TFT Serial Debug: OnUserConfirmRequired triggered... Please wait...
\//action:paused
TFT Serial Debug: OnUserConfirmRequired triggered... Heating...
TFT Serial Debug: OnUserConfirmRequired triggered... Ejecting...
TFT Serial Debug: OnUserConfirmRequired triggered... Insert and Click
echo:Insert filament and send M108
\//action:prompt_end
\//action:prompt_begin Nozzle Parked
\//action:prompt_button Continue
\//action:prompt_show
TFT Serial Debug: OnUserConfirmRequired triggered... Nozzle Parked
TFT Serial Debug: UserConfirm SD print paused done... J18
echo:busy: paused for user
echo:busy: paused for user_

I assume that line with _action:prompt_button Continue_ means that normally the button should appear.
I will dig into the functions and try to insert some prints...but the build of the fw takes always 5min...because parallel build in platform IO seems not to work.

uwetaz, 2024-05-29:
After long nights I could understand and fix it.
1) cause
The root cause is the display with its UART interface behaviour. During printing the display requests in 1.1s intervalls for 9 values (set and actual temperatures, coordinates and so on via ASCII-commands A0 to A7 and A20). The mainboard answers to the telegrams with the requested values. After that the display actualize them. Lets call them **cyclic** telegrams
If the filament sensor trips or a filament change is occured via G-Code a **acyclic** telegram or two has to be sended to the dislay (e.g. at filament sensor trip J23 to show "Lack of filament or filament monitor abnormal." with ok button which was missed sometime and J18 to show the Continue button which was also missed sometime).
I printed something without filament and not triggered filamentsensor (small filament part in the sensor) but could sporadic trip it by removing the filament part.
First I soldered wires to RX and TX of the display to have wires outside the housing during printing and tried to get the telegram history with a cheap Rigol oscilloscop and UART decoding but it was not possible to trigger to the acyclic telegrams. After that I tried to connect RX and TX to two cheap TTL UART to USB Converters (CH340G based) and visualize the telegrams via Tera Term with activated time stamps. The time stamps are not really precise but for this usage it was usefull.
I saw always the acyclic telegrams from the mainboard but not all happened in a action on display (mentioned mask or Continue button missed). I saw in error case that the time distance after the last cyclic telegram was smaller than in the working case.
I verified that with the original Anycubic FW. Here always a longer time distance between the last cyclic and the acyclic telegram was seen (I saw a minimal value of 654ms).
The pure number of characters with a display uart overflow seems not to be the root cause. It seems really to be the time distance.
2) fw change
In the fw now I use a timestamp at the last cyclic telegram (A20V) to the display and delay the acyclic telegrams until a fixed time distance is over (WAIT_MS_UNTIL_ACYCLIC_SEND). The value of 500ms worked at the first try.
3) verify the changes
I tripped 10 times the filament sensor during printing and now always the mask and Continue button occured. Without the change I got never 10 times the wished behaviour in series.
I also tested filament changes (which I originally wanted to use for hue forge printing). I provided a gcode with 10 M600 codes and now it worked perfect. The button appeared always correct.
I provide here a patch file (two files has to be changed).
I'm not really a profi in coding. Please do not hesitate to tell what could be done better.
4) additional
I also found in Marlin\src\lcd\extui\knutwurst\anycubic_touchscreen.cpp a problem with nested parentheses inside KNUTWURST_MEGA_P_LASER preprocessor part. After fixing this the compile time decreased dramatically.
If ANYCUBIC_TFT_DEBUG is active. The files from sd card are not displayed correct. One file name is showed n times depending on number files on sd card. After deactivating the define it works correct. I did not digging down to this topic.

Thanks again for the great fw (I use the great BL touch functionality).

[patch.txt](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/files/15490178/patch.txt)

knutwurst, 2024-05-29:
Wow! @uwetaz thats a great finding! Thank you so much! 

I think that because I always used the original sources from Anycubic and simply assumed that it worked that way, I never thought of observing the time offset. Great work!

I took a quick look at your patch and would implement it as it is, but it would be more elegant and understandable if you made a pull request. That way you will also appear in the contributors :)

If you don't want that, I can of course just adopt your change as it is. I leave that up to you ;)

uwetaz, 2024-05-30:
@knutwurst 
I'm on a journey until sunday. I will make the pull request probably on sunday evening.
There are sources from Anycubic? I also know the hex files.

EnableDevice, 2024-05-30:
@uwetaz, great research work! Above all praise! This approach to the task definitely commands respect. I'll try to check it on my printer a little later.

uwetaz, 2024-06-02:
Sorry I'm not finished today with the pull request. I'm not so firm with github and it is the first time to publish something in not own repo and I mixed some formal changes in the patch which I provided. I want to split in commits beside the real fix for the display for better transparency. I finish in the next days.

uwetaz, 2024-06-03:
I could now create the pull request. I hope I understood the process for PRs in github right.

knutwurst, 2024-06-05:
Added in the new [Version 1.5.4](https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/releases/tag/1.5.4) ;)

closed 2024-06-05 by knutwurst