knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S #491: Issue using pin 21 for BLTouch with V 1.5.x (same as Issue 437)

hoovercl, 2023-09-27:
I am experiencing the behavior described in issue 437, with all of the 1.5.x versions and attempting to use PIN 21 on a Mega S (I have not tried it on a Mega X, yet). It appears issue 437 was closed without any documented solution, so I am hoping it can be investigated.

I am currently running version 1.4.4 with Z_MIN_PROBE_PIN set to 21 without any issues. I have built versions 1.5.0, 1.5.1 and 1.52 with the same setting and with all three versions, when I attempt a G29, I have the same result.

[image]

I did notice one slight difference in the movement of the carriage with the 1.5.x versions compared to 1.4.4  When a G29 is sent with 1.4.4, the carriage raises, the bed moves forward, the BLTouch probe pin is deployed and then it begins probing the bed.  With the 1.5.x versions, after sending the G29, the carriage raises, and as the bed moves forward, the carriage also attempts to move to the left (doesn't occur with V1.4.4).  Afterwards, the BLTouch pin extends, then retracts, extends, retracts, and then extends a third time at which point the error message is received.

As mentioned, the error does not occur with V1.4.4.  I have confirmed this using the build I made in October 2022, and a fresh build that was created from the original download and again changing the Z_MIN_PROBE_PIN changes to 21 in pins_RAMPS.h.  The size of the .hex file from October 2022, and the one just built are exactly the same size.  

I have seen it mentioned elsewhere that this stop error is due to incorrect connection of the BLTouch to the board, however that does not seem to be the issue here since the V1.4.4 builds are working correctly.

Using the <Start Auto Leveling> from the special menu, has similar results, the only difference is the behavior of the probe which will extend, retract, extend, retract, extend, short pause, retract, extend, retract, extent, retract, followed by some tones, and then it appears the printer resets.

_Originally posted by @hoovercl in https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/issues/437#issuecomment-1705575346_

hoovercl, 2023-10-12:
I have changed the title of the issue based some further testing/information.

Initially, I did not perform the BLTouch configuration after updating the firmware, but instead updated the XYZ probe offset using the Z value determined the last time I performed the BLTouch configuration.

I was able to successfully perform all steps of the BLTouch configuration, except the final G29 step.  The commands to deploy the probe (M280 P0 S10), and stow the probe (M280 P0 S160) both worked without issue.  When the nozzle was lowered, and the probe was triggered it functioned as expected by flashing the red LED.

I have also attempted to use the G30 command to probe a single point on the bed, and this command also fails with the same error.

knutwurst, 2023-10-12:
Hi @hoovercl ,
I'm sorry that the instructions are no longer up-to-date. I'll update it.

Since version 1.5, the printer-specific pin definition has been moved to the file **pins_TRIGORILLA_14.h**. Here you can change your endstop in line 127, which is normally on 2 and in your case must be on 21.

See: 
https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/blob/e0a4e25ffa59e9e07e5446f7b9e7ca834263f00b/Marlin/src/pins/ramps/pins_TRIGORILLA_14.h#L127
I hope it helps.

Olli

hoovercl, 2023-10-14:
Hello,

Making the change as you described has corrected the issue.  Thank you very much.

I did notice that G29 seems to behave a bit differently now in that after each point is probed, the probe is retracted and then extended again after the carriage moves to the next probe point.  Is this now the normal operation?

Chris

scu00, 2023-11-23:
Yesterday I checked once again the wiring (it was right) and followed the signal from the sensor to the board to preclude a broken wire. Therefore I unplugged and re-plugged several connectors. When I tested after that, it works! No glue , why...

closed 2024-06-05 by knutwurst